raspberrypi/firmware #1050: Kernel 4.14.70-v7+ breaks f2fs filesystem

petaramesh, 2018-10-01:
Hi,
I have 2 model-3B Pis running Raspian, but on a f2fs (filesystem optimized for flash devices) root filesystem.

Both machines were _perfectly_ stable up to kernel 4.14.50-v7+

Both machines suffer random - and frequent - complete filesystem locks with kernel 4.14.70-v7+

The issue appeared immediately after upgrading from 4.14.50-v7+ to 4.14.70-v7+ on both machines.

Filesystem access locks and the machine then freezes.

So I can say for sure that “something between kernel 4.14.50-v7+ and 4.14.70-v7+ broke f2fs.

# dpkg -s raspberrypi-kernel
Package: raspberrypi-kernel
Status: install ok installed
Priority: extra
Section: kernel
Installed-Size: 114896
Maintainer: Serge Schneider <serge@raspberrypi.org>
Architecture: armhf
Source: raspberrypi-firmware
Version: 1.20180919-1
Replaces: raspberrypi-bootloader (<< 1.20160324-1)
Provides: linux-image
Breaks: raspberrypi-bootloader (<< 1.20160324-1)
Conffiles:
 /etc/default/raspberrypi-kernel aa1f56ed7333c9749dba288036f96334
Description: Raspberry Pi bootloader
 This package contains the Raspberry Pi Linux kernel.
Homepage: https://github.com/raspberrypi/firmware

Kind regards.

popcornmix, 2018-10-01:
Can you identify the exact update which caused this. See:
https://github.com/Hexxeh/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo rpi-update <hash>```
to revert back to that version. Report the first version with this error.

popcornmix, 2018-10-01:
The issue is very unlikely to be fixed if only one user is seeing a problem and the user is unwilling to help narrow down the problem.

petaramesh, 2018-10-01:
I'm happy to report that I just broke a production VPN trying to apply instructions above :-/

pelwell, 2018-10-01:
Have you ever considered having a test VPN, and only updating the production VPN with known-good configurations?

petaramesh, 2018-10-01:
On the paper, sounds like a really good idea. Such as having a spare car when mine needs repair.
In real life I only have 2 Pis on hand and a VPN that I badly need and that got quite sick last week all of a sudden after a kernel upgrade...
Not everybody works at Google. If you have a couple Pis for VPN'ing your home office, that's quite probable that you don't have any spare.

JamesH65, 2018-10-02:
Cost of Pi, $35, cost of repairing damage after failed upgrade, >$35.

Cost of car $3500, cost of repairing after incident < $3500. Not really the same thing.