raspberrypi/firmware #1055: i2c is gone after upgrade to 4.14.70

Nornode, 2018-10-07:
My i2cdetect -y 1 is gone after upgrade.

RPI: 3B 

```
uname -a
Linux 4.14.70-v7+ #1144 SMP Tue Sep 18 17:34:46 BST 2018 armv7l GNU/Linux
```

```
cat /boot/config.txt
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
gpu_mem=16
#dtoverlay=w1-gpio
dtparam=i2c_arm=on
```

pelwell, 2018-10-08:
It's working fine here:
```bash
pi@raspberrypi:~$ uname -a
Linux raspberrypi 4.14.73-v7+ #1148 SMP Mon Oct 1 16:57:50 BST 2018 armv7l GNU/Linux
pi@raspberrypi:~$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- 77
```
Are you sure nothing else has changed?
Try downgrading to an earlier version: choose a firmware version from [this page](https://github.com/Hexxeh/rpi-firmware/commits/master), make a note of the hexadecimal number on the right of the line, then run (for example) "sudo rpi-update c806ec4".

Nornode, 2018-10-08:
I'm pretty sure, yes.
I was following this guide: https://github.com/modmypi/PiModules/wiki/UPS-PIco-Installation
did an "apt upgrade" before rebooting

I have since long before [this one](https://www.modmypi.com/raspberry-pi/sensors-1061/kits-and-interface-boards-1062/r-pi-i2c-1-wire-owfs-expansion-module-) installed.

I added the picoUPS board (rev.3.0) and i2c worked fine before the upgrade (and flowing the guide).
I find it hard to believe that the install guide disables the i2c in this way?

```
lsmod | grep i2c
i2c_bcm2835            16384  0
i2c_dev                16384  0
```

Nornode, 2018-10-08:
At the moment I don't have physical access to the instance, so I'm unfortunately unable to test a fresh image.

pelwell, 2018-10-08:
I understand. I'm certain there is nothing fundamentally wrong with the I2C support in current 4.14 builds, so diagnosis is going to be a matter of finding out what is different about your installation.

Nornode, 2018-10-08:
I understand & I truly appreciate your dedication.
1. Does it in anyway matter (in the new kernel) that I have two i2c devices stacked on top of each other?
2. Any way I can revert to bcm_2708 and test if that works?

pelwell, 2018-10-08:
1. As long as its OK electrically, I doubt that a change of kernel version would make any difference.
2. Yes there is:
```
dpi@raspberrypi:~$ dtoverlay -a | grep 2708
  i2c-bcm2708
  i2c0-bcm2708
  i2c1-bcm2708
pi@raspberrypi:~$ dtoverlay -h i2c-bcm2708
Name:   i2c-bcm2708

Info:   Fall back to the i2c_bcm2708 driver for the i2c_arm bus.

Usage:  dtoverlay=i2c-bcm2708

Params: <None>
```
We switched to i2c-bcm2835 back around the 4.4 kernel so your previous kernel would have to have been very old for this to be the difference, but its worth a try.

Nornode, 2018-10-25:
I think my strange error occurred due to an error or faulty i2c device blocking also one working. Removing one got the other i2c device back to working condition. - Not a software issue.

closed 2018-10-25 by Nornode