raspberrypi/firmware #1056: kworker task blocked and not able to get_throttledlog

hpchang, 2018-10-10:
**2017-06-21-raspbian-jessie on raspberry pi 3 model B**
I use wlan0, eth0 to trigger camera module v1 and upload image periodically with my Rpi 3 B. However Rpi freezes after several hours. 

I run `vcgencmd get_throttledlog` every minute but I can't get it when it freezes. Please let me know if anything I can do for further investigation and what the better way is to ensure power always stable.

**Here are some symptoms when system freezes.**
- syslog show 
```
INFO: task kworker/0:1:3293 blocked for more than 120 seconds.
      Not tainted 4.9.28-v7+ #998
"echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
kworker/0:1     D    0  3293      2 0x00000000
Workqueue: events dbs_work_handler
[<80713b70>] (__schedule) from [<80714094>] (schedule+0x50/0xa8)
[<80714094>] (schedule) from [<80717304>] (schedule_timeout+0x220/0x364)
[<80717304>] (schedule_timeout) from [<80714c08>] (wait_for_common+0xbc/0x17c)
[<80714c08>] (wait_for_common) from [<80714ce8>] (wait_for_completion+0x20/0x24)
[<80714ce8>] (wait_for_completion) from [<805d92d4>] (rpi_firmware_transaction+0x68/0xac)
[<805d92d4>] (rpi_firmware_transaction) from [<805d9410>] (rpi_firmware_property_list+0xf8/0x220)
[<805d9410>] (rpi_firmware_property_list) from [<805d959c>] (rpi_firmware_property+0x64/0x84)
[<805d959c>] (rpi_firmware_property) from [<805b4888>] (bcm2835_cpufreq_clock_property.constprop.1+0x48/0x5c)
[<805b4888>] (bcm2835_cpufreq_clock_property.constprop.1) from [<805b48e8>] (bcm2835_cpufreq_driver_target_index+0x4c/0xc0)
[<805b48e8>] (bcm2835_cpufreq_driver_target_index) from [<805aeb5c>] (__cpufreq_driver_target+0x250/0x580)
[<805aeb5c>] (__cpufreq_driver_target) from [<805b3034>] (od_dbs_timer+0xa8/0x174)
[<805b3034>] (od_dbs_timer) from [<805b3cac>] (dbs_work_handler+0x3c/0x68)
[<805b3cac>] (dbs_work_handler) from [<80136594>] (process_one_work+0x158/0x454)
[<80136594>] (process_one_work) from [<801368f4>] (worker_thread+0x64/0x568)
[<801368f4>] (worker_thread) from [<8013c7fc>] (kthread+0x10c/0x124)
[<8013c7fc>] (kthread) from [<80108148>] (ret_from_fork+0x14/0x2c)
```
- ssh cannot login
- The red indicator of camera keeps lighting
- cron job sometime are still working and able to send some log info back to server

**Here are some workaround result to recovery system freeze.**
- cron job running cmd `reboot` doesn’t recover it. Only power cycle can 100% recover it
- Sometime cron job running cmd `echo b > /proc/sysrq-trigger` could recover it
- Under voltage of power adapter impacts this issues. For very bad power adapter, this issue happens more frequently. 

**The following debug info is logged after system freezes 1 hr.** 
- Test starts on 17:50
- System freezes on 19:00
- Log debug info on 20:00 as follows
    - vcgencmd version
    - vcgencmd get_config int
    - vcgencmd get_config str
    - vcdbg log msg
    - vcdbg log assert
    - vcdbg reloc
    - vcdbg malloc
- Every minute to log voltage status after boot. Not sure under voltage info when system freezes.
    - vcgencmd get_throttled

- syslog summary
```
Sep  5 17:50:02 raspberrypi kernel: [    0.000000] Booting Linux on physical CPU 0x0
................................SKIP MANY LINES................................
Sep  5 18:57:01 raspberrypi CRON[5875]: (root) CMD (sudo python /home/pi/code/RPi/pull.py)
Sep  5 18:58:01 raspberrypi CRON[5890]: (root) CMD (sudo python /home/pi/code/RPi/pull.py)
Sep  5 18:59:01 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Wed Sep  5 19:00:31 2018 [try http://www.rsyslog.com/e/2007 ]
Sep  5 18:59:01 raspberrypi CRON[5906]: (root) CMD (sudo python /home/pi/code/RPi/pull.py)
Sep  5 18:59:49 raspberrypi kernel: [ 4175.853799] INFO: task kworker/0:1:3293 blocked for more than 120 seconds.
Sep  5 18:59:49 raspberrypi kernel: [ 4175.853811]       Not tainted 4.9.28-v7+ #998
Sep  5 18:59:49 raspberrypi kernel: [ 4175.853816] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Sep  5 18:59:49 raspberrypi kernel: [ 4175.853822] kworker/0:1     D    0  3293      2 0x00000000
Sep  5 18:59:49 raspberrypi kernel: [ 4175.853850] Workqueue: events dbs_work_handler
Sep  5 18:59:49 raspberrypi kernel: [ 4175.853883] [<80713b70>] (__schedule) from [<80714094>] (schedule+0x50/0xa8)
Sep  5 18:59:49 raspberrypi kernel: [ 4175.853899] [<80714094>] (schedule) from [<80717304>] (schedule_timeout+0x220/0x364)
Sep  5 18:59:49 raspberrypi kernel: [ 4175.853915] [<80717304>] (schedule_timeout) from [<80714c08>] (wait_for_common+0xbc/0x17c)
Sep  5 18:59:49 raspberrypi kernel: [ 4175.853932] [<80714c08>] (wait_for_common) from [<80714ce8>] (wait_for_completion+0x20/0x24)
[... 1067 lines cut ...]
0x3eb5f7c0 = malloc(740)  [mmal_pool payload]
0x3eb5fae0 = malloc(740)  [mmal_pool payload]
0x3eb5fe00 = malloc(740)  [mmal_pool payload]
0x3eb60120 = malloc(48)   [MMAL pool]
0x3eb60180 = malloc(4.7K) [MMAL buffer headers]
0x3eb61440 = malloc(764)  [mmal ril omx buf pool]
0x3eb61760 = malloc(764)  [mmal ril omx buf pool]
0x3eb61a80 = malloc(48)   [MMAL pool]
0x3eb61ae0 = malloc(480)  [MMAL buffer headers]
0x3eb61ce0 = malloc(80)   [mmal ril omx buf pool]
0x3eb61d60 = malloc(64)   [call from 0x3df13d0e]
0x3eb61dc0 = malloc(456)  [LSACLS Thread Data]
0x3eb61fc0 = malloc(128K) [LSACLS Stack]
0x3eb81fd0 = hole(4.0M)
Malloced:5.7M Remaining:4.0M Failed allocs:0
```
- vcgencmd get_throttled (every 1 min) and gets 61 times until RPi freezes.
```
throttled=0x0
......GET 61 times......
throttled=0x0
......NO more info after system freezes......
```

It seems to me that power is critical to this issue. However I cannot confirm `vcgencmd get_throttled` when issue happens.

hpchang, 2018-10-11:
@6by9 Thank you for your suggestion. Should I upgrade `from Jessie to Stretch` **or**  just do `sudo apt-get update` + `sudo apt-get dist-upgrade`?

JamesH65, 2018-10-11:
Jessie to Stretch can be awkward. If you can simple create a new SD card with the latest OS, then move your data across, that would be the most reliable solution.

6by9, 2018-10-11:
```apt update``` ```apt  dist-upgrade``` will only upgrade packages within the distribution, ie whatever has been released for Jessie. As I said, Jessie has not been maintained since Stretch has been out and stable, therefore it won't gain you anything.

Upgrading between major OS steps can be done but is not advisable. As James says, it is safest to start again from scratch with a Stretch install.

monanto, 2019-01-17:
Having same issue?

You find the solution ?

hpchang, 2019-01-21:
@monanto We find that unstable power reproduces this issue more easily. After replacement of power and cable, this issues seems to be fixed after running over one month.

monanto, 2019-01-21:
Ok

thanks