raspberrypi/firmware #1057: MMAL image_fx with advanced deinterlacing breaks audio/video sync

setmind, 2018-10-14:
From commit a46b1f9 on Mar 14, 2018, I can no longer get audio and video in sync with any 1080i60 source deinterlaced by image_fx in the advanced mode.  This was working fine up to commit 0f5f899 on March 13, 2018.  I am using Raspberry Pi 2 Model B Rev 1.1, and am experiencing this problem with both a custom program and omxplayer.

Here are commands to reproduce the problem:
Video lags:
`omxplayer 20090227_233739-H.264.m2ts.mp4`
A/V sync is fine:
`omxplayer --advanced=0 20090227_233739-H.264.m2ts.mp4`
A/V sync is fine:
`omxplayer --nodeinterlace 20090227_233739-H.264.m2ts.mp4`

The sample video is available here: https://www.hdpvrcapture.com/wordpress/?page_id=97

Did anything change around QPU?

6by9, 2018-10-15:
Minor correction - a46b1f9 was MAY, not March.

The changelist is complete between those two commits for all firmware changes.
I haven't checked the kernel as that is GPLv2 and source is available anyway.

setmind, 2018-10-17:
My bad.  I entered wrong commits.  Here are the right ones:

Mar 13, 2018, 3347884 - bad
Mar 9, 2018, 6e3c5f8 - ok

It seems that from the Mar 13 commit, v3d clock is stuck at 250 MHz (modification to config.txt has no effect).  This explains video lag, but I am not sure why this is happening.

setmind, 2018-10-17:
Thanks.  That works.  Now I see v3d clock increases to 300 MHz during video playback and see no video lag.

popcornmix, 2018-10-17:
Yes, I'm afraid that's the new default.
Pi3 gets auto_turbo my default (as the manufacture testing tests the core at higher frequencies).
Pi1/2 has auto_turbo disabled by default, but can be overridden in config.txt as that testing wasn't done.
However I haven't seen a Pi that fails with the auto turbo frequencies, so the vast majority should be fine.

setmind, 2018-10-18:
Thanks.  That explains the problem.  Closing this issue.

closed 2018-10-18 by setmind