raspberrypi/firmware #1061: Make firmware updates available via the lvfs project

lhirlimann, 2018-10-16:
Would be nice to be able to apply firmware updates using fwupdmgr. This would support most Linux distro. Documentation is at https://fwupd.org/vendors.

pelwell, 2018-10-16:
The second sentence says (their emphasis):

> This service should only be used to distribute files that are _flashed_ onto non-volatile memory.

One could choose to read that sentence in a way that encompasses a Pi firmware download, but I doubt that would be the intended interpretation.

6by9, 2018-10-17:
> There are currently 309 devices supported on the LVFS with 941 available firmware versions.

So approx 3 versions of firmware per device. So either we end up swamping their servers, or only update once in a blue moon.

It looks very much like it is intended for things like the firmware in a webcam, motherboard BIOS, or other device that is infrequently updated and uses a firmware upload type mode to reprogram the onboard flash device.

pelwell, 2018-10-17:
It's not going to happen.

closed 2018-10-17 by pelwell