raspberrypi/firmware #1063: disable_touchscreen=1 completely turn off the display

stephanedamo, 2018-10-22:
Hello,
On the latest kernel versions, setting "disable_touchscreen=1" in config.txt completely disables the screen (the backlight won't turn on) instead of just disabling the touch. Tested on the official 7" display.
I've tried another kernel version from 2 months ago and it works as expected.

6by9, 2018-10-23:
I suspect it is the backlight change in https://github.com/Hexxeh/rpi-firmware/commit/e0a0ba809ef11f6bed8a00d34dda961239ea432c
Can you try ```rpi-update 9f23092``` to get the version before to confirm?

Ping @JamesH65 - we were talking about this option only yesterday.

6by9, 2018-10-23:
Reproduced this on the currently firmware. Yes the backlight callback gets set as the touchscreen thread starts, so with the delay to setting the backlight we end up with it totally off.

James has made further cleanups to the backlight control which resolve this. They've been merged to our internal tree but not released yet. I'd expect them in the next rpi-update.

(This does raise the issue that with this there is now a mechanism where the VPU can be using i2c-0 and in conflict with the ARM. I take the view that we'll just ignore that case for the time being).

stephanedamo, 2018-10-23:
Thanks ! I was trying this option because I'd like to talk to the touchscreen controller on I2C-0 directly without the GPU interfering. I'm not sure using this option is enough to avoid conflicts though. If it's a hardware issue, i'll live with it.

6by9, 2018-10-23:
Ah, useful context as that was the conversation I had with James yesterday.

The issue is that there is no arbitration between the ARM cores and the VPU (the VideoCore Vector Processing Unit). Should they both try I2C transactions over the interface at the same time then "Bad Things" happen.
When an interrupt occurs it'll kick both processors, and whichever gets there first (typically the VPU) will try servicing it. If it was a transaction for that processor then all is OK. If it was a transaction for the other processor then the even gets cleared as unexpected, and when the other processor gets around to servicing the interrupt there's nothing to find.

If you can ensure that you never update the backlight level whilst trying to read the touchscreen then you should be OK (ie let it be set at boot and never touch it).
Likewise you can not run the camera as that uses i2c-0.

stephanedamo, 2018-10-23:
Mmh so it should be doable since I'm doing everything in bare-metal with the Circle lib, without camera and backlight control. Do you know the slave address of the touch controller on I2C-0 ?

6by9, 2018-10-23:
I believe the touchscreen is on address 0x38. The Atmel chip that controls the DSI/LVDS bridge and backlight is on 0x45.

popcornmix, 2018-10-23:
@stephanedamo the firmware update @6by9 referenced is now in latest rpi-update firmware.
Can you check if the issue with ``disable_touchscreen=1`` is resolved now?

stephanedamo, 2018-10-23:
Problem fixed ! Thanks for the reactivity.

closed 2018-11-03 by stephanedamo

MrMoins, 2019-08-21:
I found that I cannot manage backlight (brigthness, power) if I disable the touchscreen (`disable_touchscreen=1` in `/boot/config.txt`)

`/sys/class/backlight/` is empty when `disable_touchscreen=1`

it's really weird because I thought it only turned off the touch, not the total control of the screen.

```
# cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
...
# uname -a
Linux rds 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
```