raspberrypi/firmware #1066: DeviceTree bt_pins available also with disabled Bluetooth

pvizeli, 2018-10-27:
I use for HassOS a systemd service file like above (that is only an example). Since 4.14.32 the condition doesn't work anymore because the overlay doesn't remove this folder with `dtoverlay=pi3-disable-bt`. Exists a new way to evaluate the Bluetooth state?

Project: https://github.com/home-assistant/hassos

```
[Unit]
Description=Configure Bluetooth Modems connected by UART
ConditionPathIsDirectory=/proc/device-tree/soc/gpio@7e200000/bt_pins
Before=bluetooth.service
After=dev-ttyAMA0.device

[Service]
Type=forking
ExecStart=/usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow -

[Install]
WantedBy=multi-user.target
```

pelwell, 2018-10-29:
As far as I can remember, this test has never been able to identify a Pi with disabled Bluetooth - it is intended to avoid running the script on Pis without any Bluetooth capability. Even Raspbian attempts to start the hciuart service and fails - this isn't something we've tried to avoid up to now, but it is relatively simple to change that.

The rpi-4.14.y branch of the kernel source now includes an updated pi3-disable-bt overlay that shrinks the properties under bt_pins to zero length (it isn't possible to delete them, or to delete the bt_pins node itself). To take advantage of this change there's a [Pull Request against the hciuart service](https://github.com/RPi-Distro/pi-bluetooth/pull/7) that changes the condition to check that the `brcm,pins` property pseudo-file is non-empty.

pvizeli, 2018-10-29:
Thanks for your support 👍

closed 2018-10-29 by pvizeli