raspberrypi/firmware #1069: gpio-poweroff increases current consumption

onemikedelta, 2018-11-14:
PI Zero W wastes huge amount of power when "dtoverlay=gpio-poweroff,gpiopin=16,active_low=1" used in the config file. Pin is working as expected.
My measuremets are here:
Running on Idle :100mA  
Turned off without gpio-poweroff: 30mA
Turned off with gpio-poweroff: 190mA

pelwell, 2018-11-14:
That's expected behaviour. The gpio-poweroff mechanism prevents the ARM from shutting down - instead it sleeps waiting for the external power to be switched off as a result of the transition on the GPIO.

What's your use case?

onemikedelta, 2018-11-14:
My system runs with Li-Ion batteries and a small ups module. If I can decrease current below 50mA ups module automatically cuts off power. To achieve this low power I tried to turn off my hat using that gpio pin.
I expected to see some current drown but isn't it abnormal?

onemikedelta, 2018-11-14:
It is 100mA. When processing it is between 140-170 mA. As you can see this gpio-poweroff tool draws current more than processing. I think there is a bug like infinite loop etc. or maybe misconfiguration and something is shorting internally in the soc. I am not familiar with the arm core but if it is the expected current it is better to leave the system on than turning off to save power. Then whole thing became useless. This makes me think that there is a bug.

pelwell, 2018-11-14:
It's not really a bug - you are using the gpio-poweroff module in a way it wasn't intended, causing the ARM to return from a function that shouldn't return and end up spinning in a dead end. There is a `timeout-ms` property that would increase the time it waits before spinning (the default is 3 seconds), but the current draw may still be too high with the ARM on. You can download an updated overlay with a `timeout_ms` parameter [from here](https://drive.google.com/file/d/1zw21tOzePzU5LpFTtWFcGLZyvu9a5UWr/view?usp=sharing). To set the delay to 20 seconds use:
```
dtoverlay=gpio-poweroff,gpiopin=xxx,timeout_ms=20000
```

It would be simpler if you could use the GPIO to tell the UPS to switch off, but I assume that's not possible. As an alternative, you could pick a GPIO which naturally pulls in the direction that turns off the HAT, and use the `gpio=` config.txt setting to drive against the pull and keep the HAT on. When the Pi shuts down the drive should be cancelled and the HAT should turn off. For example, if you chose GPIO 12 which naturally pulls low, your config.txt entry would say:
```
gpio=12=op,dh
```

JamesH65, 2019-01-08:
@onemikedelta Any progress or further comments on this?

onemikedelta, 2019-01-08:
@JamesH65 When the last time I tried, it was still the same(~15 days ago). Bug still exists. 
I give up trying because couldn't find a solution. It continuously draws power and prevents the ups module to turn off system. If someone else tries to make this kind of auto power off system, they should use a mosfet etc. to cut power.

closed 2019-11-05 by JamesH65