raspberrypi/firmware #1070: RPi 2B : Wifi USB dongle prevents boot on USB storage

timcrouzet, 2018-11-17:
Hi,

I have a RPi 2B : 
SD card with latest bootcode.bin (timeout and UART empty files)
SSD pluged via USB adapter

**Boots fine on the SSD :** 
```
Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 1
Trying USB
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 5 ports, multi_tt = 1
Setting interface 0
Enabling PORT POWER on port 1
Enabling PORT POWER on port 2
Enabling PORT POWER on port 3
Enabling PORT POWER on port 4
Enabling PORT POWER on port 5
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Device found: type = Ethernet adapter, addr = 5
Found device on port 2
Found highspeed device
Device found: type = Mass storage, addr = 6
Trying booting from MSD device addr 6
Retrieving the config descriptor
Found ep_in = 00000001, ep_out = 00000002
GetMaxLun returned 0, max_lun = 0
Sending and retrying TEST_UNIT_READY
Sending SCSI_COMMAND_INQUIRY
Sending READ_CAPACITY
init_msd returned 0
USB device configuration set
Initialised mass storage, block size = 512, block count = 46905264
MBR read and decoded successfully
FAT32 partition decoded successfully
start.elf found in first partition on USB MSD, booting
Read File: config.txt, 1609 (bytes)
```

**But when I add my wifi dongle it fails to boot :**
```
Found SD card, config.txt = 0, start.elf = 0, recovery.elf = 0, timeout = 1
Trying USB
Hub device found at addr 4, enumerating HUB
Initialise hub
Found 5 ports, multi_tt = 1
Setting interface 0
Enabling PORT POWER on port 1
Enabling PORT POWER on port 2
Enabling PORT POWER on port 3
Enabling PORT POWER on port 4
Enabling PORT POWER on port 5
Waiting for devices to respond to reset
Found device on port 1
Found highspeed device
Device found: type = Ethernet adapter, addr = 5
Found device on port 4
Found highspeed device
Device found: type = Ethernet adapter, addr = 6
Trying booting from Ethernet device addr 5
Initialise ethernet with MAC b8:27:eb:49:79:9e
Wait for Link up
Link up
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
```

Looks like it does not detect the SSD and tries to boot from network :(

My dongle is a TPLink TL-WN725N (r8188eu chip)

Here is a lsusb (when dongle plugged after boot) 
pi@raspberrypi:~$ lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/5p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M
        |__ Port 2: Dev 4, If 0, Class=Mass Storage, Driver=usb-storage, 480M
        |__ Port 4: Dev 5, If 0, Class=Vendor Specific Class, Driver=r8188eu, 480M

Thanks very much for your help!

pelwell, 2018-11-17:
Does the power LED flicker at all during the boot process, with and without the WiFi dongle?

timcrouzet, 2018-11-17:
Not at all, the red power led is perfectly stable in both cases.

pelwell, 2018-11-17:
Do you get different results with the external devices on a hub (ideally powered)?

ghollingworth, 2018-11-18:
That's interesting, I can only assume it's timing out quicker when you insert the dongle and therefore not waiting for the SSD...

I wonder if we can add an option to force it to wait for a specific number of devices, to make sure it won't time out early...

timcrouzet, 2018-11-19:
Thanks all for your answers. On pelwell's recommendation I got ahold of a powered USB hub and it solved the issue!!

What is weird is that a hot plug of the wifi dongle was fine. But apparently at boot time it was not for the SSD

closed 2018-11-19 by timcrouzet