raspberrypi/firmware #1072: Memory split incorrect after updating firmware blobs and kernel

dflogeras, 2018-11-22:
I am testing a more recent kernel to see if it fixed a problem that crept into the camera interface in the last couple tagged kernels.  The camera is not what I'm discussing here.

The hardware is a RPi Zero-W with 512MB.  Typically I have in my config.txt:
gpu_mem_512=64
start_x=1

but I also had left start_debug=1 set from a previous camera problem.

Under the last released kernel that works with my camera, 4.14.62-g1f89ad77bf9b, this results in roughly 448MB of ram left for Linux, as expected.

I tried https://github.com/raspberrypi/linux/commit/f6878de4d2818c2256b2a340eaeeabfb0b24c71b and then I also upgraded my firmware blobs to https://github.com/raspberrypi/firmware/commit/55e591283f456ab9cebd9c31aaef4939f814880d

My camera issue seems to have been fixed, but I noticed that it is no longer splitting my memory as I have asked. After upgrading to the firmware/kernel stated above, and rebooting, I am left only with 128MB of system ram.  If I comment out start_debug=1 in my config.txt, I get what I requested.

As a sanity check, I booted back into the old kernel, with start_debug=1 and it works as expected (448CPU/64GPU).

pelwell, 2018-11-22:
128MB of RAM is what the Pi will appear to have if you have chosen a start*.elf that doesn't have a matching fixup*.dat. By matching I mean it is for the same variant (start.elf->fixup.dat, start_x.elf->fixup_x.dat, etc.) and both were from the same firmware release (assuming the release isn't broken).

dflogeras, 2018-11-23:
PEBKAC

Thanks for the gentle nudge, and sorry for the noise.

closed 2018-11-23 by dflogeras