raspberrypi/firmware #1074: vc_dispmanx_resource_write_data takes a long time

ktb92677, 2018-11-27:
Hi all! I set up a pretty basic while loop based on hello_pi/hello_dispamx. It looks as follows:

while (true) {
   copy(frame_buffer, frame_buffer + sizeof(frame_buffer), openmax_buffer);
   vc_dispmanx_resource_write_data(resource, VC_IMAGE_ARGB8888, width * 4, openmax_buffer, dst_rect);
}

This basic code structure does work. However the call to vc_dispmanx_resource_write_data is much MUCH too slow taking about 20+ms per render cycle. Why is it taking so long?

JamesH65, 2018-11-27:
You haven't said how big the buffer is. which will have a direct influence of the time taken I believe.

ktb92677, 2018-11-27:
> You haven't said how big the buffer is. which will have a direct influence of the time taken I believe.

The buffer size is as follows: 1920 * 1080 * 4

ktb92677, 2018-11-27:
> Your variable `openmax_buffer` sounds suspiciously like you're mixing and matching OpenMax IL and DispmanX.
> Is there a reason for not using the IL video_render component. Particularly when tunnelled it avoids all the copying to/from GPU memory.

Hmm, I didn't know IL was capable of drawing raw ARGB framebuffers... do you think you could link me to an example?

6by9, 2018-11-27:
> Hmm, I didn't know IL was capable of drawing raw ARGB framebuffers... 

I was more taking your buffer name and making guesses. What are you doing with IL? Where/what is creating the source buffer? Describe your use case.

```VC_IMAGE_ARGB8888``` is equivalent to ```OMX_COLOR_Format32bitARGB8888```.

Depending on your use case you're actually better off with MMAL as there there is a zero copy path. Both IL and DispmanX have an inherent concept of host vs GPU buffers, and copying everything between them.

ktb92677, 2018-11-27:
Here's my use case: Currently I am using IL to render h264 frames to the screen and it is working great. What I wanted to do is render something ontop of my h264 stream that has opaque and transparent portions. Using dispmamx is working, but it is consistently taking nearly 30ms every render cycle to update the screen, which is much much too slow.

6by9, 2018-11-30:
https://github.com/6by9/mmal_render_example gives a simple MMAL rendering example, although using RGB24 instead of RGBA. Change ```ENCODING``` appropriately (both MMAL_ENCODING_BGRA and MMAL_ENCODING_RGBA are supported by video_render).

I'll add a commit on top that enables zero copy, which should give a performance boost as you end up with a cache flush/invalidate instead of copying the data.

ktb92677, 2018-11-30:
> https://github.com/6by9/mmal_render_example gives a simple MMAL rendering example, although using RGB24 instead of RGBA. Change `ENCODING` appropriately (both MMAL_ENCODING_BGRA and [...]

I actually posted an issue on that example. I tried to set it to RGBA and the program throws an error. As for the zero copy commit, will that take much time?

6by9, 2018-11-30:
> I actually posted an issue on that example. I tried to set it to RGBA and the program throws an error.

Responding on that issue. Check the supported formats.

> As for the zero copy commit, will that take much time?

No, it's done (one line change).

ktb92677, 2018-11-30:
> > I actually posted an issue on that example. I tried to set it to RGBA and the program throws an error.
> 
> Responding on that issue. Check the supported formats.
> 
> > As for the zero copy commit, will that take much time?
> 
> No, it's done (one line change).

So... 32 bits per pixel is not supported? Because I tried many combinations and they all returned a similar error.

6by9, 2018-11-30:
As per response on https://github.com/6by9/mmal_render_example/issues/2, you said you set MMAL_ENCODING_RGB32, not RGBA (or BGRA)

I've checked that MMAL_ENCODING_RGBA and  MMAL_ENCODING_BGRA work in that app.

ktb92677, 2018-11-30:
Okay that seems much much faster. This issue can be closed out

closed 2018-11-30 by ktb92677