raspberrypi/firmware #1075: Noobs boots and installs but no system startup - Kingston 120gb SSD

puccaso, 2018-11-28:
Just following the request via https://www.raspberrypi.org/blog/pi-3-booting-part-i-usb-mass-storage-boot/

I used etcher initially, to burn the raspberry-desktop img to the SSD, but no avail. didn't boot.
So i went with a fat fs to copy Noobs over directly, and hey presto! got a long rainbow screen, but i gather it was resizing the drive/image.

the system installed completely, however the system does not boot. 
I'm just stuck, red LED and nothing else. 

Any ideas?

puccaso, 2018-11-28:
edit: ive copied the the bootcode.bin and still no avail.
there are two partitions currently, boot and recovery 
i copied the bootcode to both, on separate occasions and still no boot

JamesH65, 2019-01-08:
@puccaso Did you get anywhere with this or are you still stuck? Have you checked the documentation here https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md

puccaso, 2019-01-08:
Hey

Thanks for reaching out.

I've gone through all of that.. I just put a hack in place, using the SSD
as root but still booting from the SD but no luck otherwise.

I thought that the usb3 might be a problem, but .. noobs booted and
installed, so why the installed system didn't boot is beyond me.