raspberrypi/firmware #1076: start_file doesn't work inside config.txt filters

ali1234, 2018-11-29:
Example config.txt:

```
[all]
gpio=5=ip,pu

[gpio5=0]
start_file=msd.elf

[gpio5=1]
start_file=start.elf
```

Expected result: Pi should boot msd.elf if gpio5 is externally pulled low.

Actual result: Pi boots whatever start_file is last in the file, regardless of filters.

popcornmix, 2018-11-30:
Filtering is only supported by start.elf, not bootcode.bin (which is much simpler).
So, filtering won't work on config.txt options handled by bootcode.bin. e.g.
```
gpu_mem
gpu_mem_256
gpu_mem_512
gpu_mem_1024
total_mem
require_total_mem
startx
start_x
start_debug
boot_partition
legacy_mapping
bootcode_delay
force_pvt
uart_2ndstage
start_file
fixup_file
```

popcornmix, 2018-12-07:
I've had a go at adding gpio filters to bootcode.bin. Can you try this bootcode.bin?
https://drive.google.com/uc?id=19LE7WHmbBZgxEowi48Q8ilOVeg0nQrm7&export=download

ali1234, 2018-12-07:
I tested this using rpi-ramdisk. The msd/start.elf is renamed to msd.elf and added alongside all the usual firmware files. Then my config.txt is similar to the above but loading start_x.elf and my initrd in the normal case:

```
gpio=5=ip,pu

[gpio5=0]
start_file=msd.elf

[gpio5=1]
start_x=1
initramfs initrd
```

I tested this with rpiboot on a Pi Zero and copying everything to the SD card on a 3A+. In both cases it worked - the normal firmware would boot unless gpio 5 is pulled low, and then the device goes to MSD mode where I can update the files on the SD card.

This is a HUGE time saver when I am testing initrds as I no longer need to put the SD card into a PC to reflash after every build. For Zero I could use rpiboot and for the 3B I could use PXE booting. But the 3A+ current can't do either of those so this is a great help.

Are there any particular corner cases you would like me to test?

JamesH65, 2019-01-08:
@popcornmix Did this get merged? OK to close?

popcornmix, 2019-01-08:
Yes, this is in rpi-update bootcode.bin (pushed 22 days ago).

JamesH65, 2019-01-08:
Closing this issue as questions answered/issue resolved.

closed 2019-01-08 by JamesH65