raspberrypi/firmware #1077: hello_triangle2, "vertex" attribute of mandelbrot program is not setup

kaomoneus, 2018-11-29:
Namely state->attr_vertex2 is always unitialized.
And it perhaps works due to bug in gles2 implementation, somehow it get linkage for the same-name attribute from another (julia) shader.

kaomoneus, 2018-11-29:
Please find a patch in attachment.
--Cheers!

[attr_vertex2-issue-1077.patch.txt](https://github.com/raspberrypi/firmware/files/2630417/attr_vertex2-issue-1077.patch.txt)

6by9, 2018-11-29:
Please create a Pull Request against https://github.com/raspberrypi/userland/tree/master/host_applications/linux/apps/hello_pi. Userland is the source of these files.

JamesH65, 2019-01-09:
I beleive this has been done so closing.

https://github.com/raspberrypi/userland/pull/508

closed 2019-01-09 by JamesH65