raspberrypi/firmware #1079: disable_overscan is not having any effect in current raspbian firmware

hhromic, 2018-12-05:
According to documentation [1], the `disable_overscan` option should disable overscan processing on the GPU, however setting it to `1` doesn't seem to do anything.

If I set the `overscan_{left,right,top,bottom}` options to visible values such as `40`, I get the expected black borders around the framebuffer when `disable_overscan` is commented or `0`, however setting it to `1` doesn't remove the black borders as expected. The GPU is still processing an overscan.

I'm using a computer monitor over HDMI for testing and it is using DMT for video mode. Maybe the `disable_overscan` option doesn't work on DMT modes? shouldn't be the case.

Also, even after setting `disable_overscan` to `1`, `vcgencmd` still reports the option to be `0`, making me believe it's being ignored by the firmware.

[1] https://www.raspberrypi.org/documentation/configuration/config-txt/video.md

popcornmix, 2018-12-05:
``disable_overscan=1`` removes the *default* overscan settings.
These are:
48 pixels on each edge for HD CEA modes
32 pixels on each edge for SD CEA modes
0 for DMT modes (which displays don't overscan).

You can still apply additional overscan settings which are added to the default.

So the behaviour you describe is as expected.

hhromic, 2018-12-05:
A-ha! So, in reality, `disable_overscan=1` is actually just a convenience shortcut for setting `overscan_{left,right,top,bottom}=0` ?
If so, then yes, that would explain all the behaviour I have been experiencing and is not a bug indeed.
However, I think the documentation should be updated then to better explain what this option actually does.

I was mislead on thinking that the setting disables (as its name implies) all sort of overscan processing, hence I expected that even setting the individual margins wouldn't matter. If I understand well, overscan is always applied (just with 0 pixels when "disabled"), does this mean there is always a penatly on scaling the video output? Or this is only costly with `overscan_scale=1`? 

Thanks in advance for all the clarifications.

popcornmix, 2018-12-05:
Overscan (without ``overscan_scale=1``) doesn't cause any additional scaling - the framebuffer size is reduced by the amount of overscan, so pixels are unscaled.
``overscan_scale=1`` is generally not recommended and will result in elements being scaled.

hhromic, 2018-12-05:
Thanks for the clarification! I will close this issue now.
If you agree that the documentation should be improved for the `disable_overscan` setting (to explain it actually sets the four other margins to `0`), let me know and I can send a PR for that.

closed 2018-12-05 by hhromic

hhromic, 2018-12-05:
@JamesH65 submitted now, thanks!