raspberrypi/firmware #1080: How to download kernel 4.9.80 Raspbian 2018-03-13 sources?

rodizio1, 2018-12-12:
I'm facing a problem I don't really understand, could somebody please shed some light into this and help me downloading Kernel 4.9.80 sources?

Usually, when I want to have exactly the same kernel (but from source) as in a particular Raspbian Release, I just use Notro's rpi-source (https://github.com/notro/rpi-source), which somehow pulls the correct kernel sources by looking at some commit ID in the Rasperrry Pi firmware package. This so far worked for all Raspbian releases I tried during the last 2 1/2 years. 

But not for Raspbian 2018-03-13, there, it pulls 4.14.24, which is not the correct Kernel version for Raspbian 2018-03-13.

I opened an issue in Notro's rpi-source repository here: https://github.com/notro/rpi-source/issues/40 and Notro also confirmed that the fw commit points to the wrong kernel version.

Interestingly, there is a changelog entry in the raspberrypi-firmware package changelog file, that (atleast how I interpret it) states exactly that:

>     raspberrypi-firmware (1.20180313-1) stretch; urgency=medium
> 
>         firmware as of 3347884c7df574bbabeff6dca63caf686e629699
>         Kernel 4.9.80
>             Breaks utilities which use firmware commit to determine kernel commit
> 
>     -- Serge Schneider serge@raspberrypi.org Tue, 13 Mar 2018 21:05:21 +0000
> 

So it appears that this is not an issue with rpi-source, but with the changelog file (?).

I have two questions:
- How can I download the correct 4.9.80 kernel version (preferrably as tar.gz, i.e. without having to clone the whole raspberry linux kernel via Github)
- Out of interest: Why is the commit ID pointing to a different version?

(not sure if this qualifies as an issue here, sorry for the noise if it doesn't)

popcornmix, 2018-12-12:
@XECDesign is this something you know about?

XECDesign, 2018-12-14:
I can't check right now, and don't recall the specifics off the top of my head.

It looks like we had to ship an earlier version of the kernel than what was in the firmware repo.  We may have had a bug before release and had to roll back the kernel and maybe we didn't have a stable branch yet to use for this purpose. Or we had to ship the earlier kernel with a patch that couldn't be made public yet.

JamesH65, 2019-01-22:
ping @XECDesign

XECDesign, 2019-01-22:
pong? I can't do anything about this.

JamesH65, 2019-01-22:
Is this for closing then?

XECDesign, 2019-01-22:
I believe @popcornmix would've built it from a non-public branch at the time, but there should be a corresponding commit now. I just don't know what it is.

popcornmix, 2019-01-22:
My guess is that we needed a firmware bump for raspbian (for a critical bug fix or to support a new product) when apt was still on 4.9 and firmware repo (and rpi-update) was testing 4.14.

So I guess the apt update took the firmware from 3347884c7df574bbabeff6dca63caf686e629699 and the kernel from an earlier commit (looks like b1a7f4aea6cbd380319c2849ecc5988f9a4dba70)

So you want to run rpi-source from b1a7f4aea6cbd380319c2849ecc5988f9a4dba70.
I'm not very familiar with this, but it appears to use ``/boot/.firmware_revision`` to get the revision to use, so
```
echo b1a7f4aea6cbd380319c2849ecc5988f9a4dba70 | sudo tee /boot/.firmware_revision
rpi-source
```
might be what you want.

rodizio1, 2019-01-23:
Thanks a lot, I'll give that a try.

closed 2019-11-05 by JamesH65