raspberrypi/firmware #1081: 3B+ PXE boot broken

SairesArt, 2018-12-12:
I'm booting a 48 PI's from a x86 Raspbian PC. The PI's get their power from the PC's PSU and are thus turned on the same time the PC boots.
However, because the PC takes 40 secs to boot the PI's first bootpc requests go into the dark. The problem is, although I read that the PI's try to network boot forever, they never actually continue requesting. As a result the PI's do not network boot when powered on at the same time as the x86 PC.

The PI's properly issue their bootpc requests if connected to power after the x86 system booted and that works (because the very first request is answered). I got told that this is the result of a firmware bug and although the PI's are supposed to issue boot requests, they never actually issue more than 4 or so. 

Can this be somehow circumvented? (without having to plugin 48 SD cards)
Can somehow maybe a DHCP answer from another source help them to stay alive?
I got suggested to wire up all the hardware reset pins and trigger them after system boot, which is currently the only proper solution as far as I understand.

ghollingworth, 2018-12-13:
You're right the bootcode is broken in this situation, the only way to fix it is to either power off / on again or use the reset pin as noted...  Other possibility is to connect all the power supplies through an Energenie power switch and then get the PC to power them on when it has booted...

SairesArt, 2018-12-14:
> 
> 
> You're right the bootcode is broken in this situation, the only way to fix it is to either power off / on again or use the reset pin as noted...

Ahh dammit. Not a deal breaker, but this should be seriously noted in the [official tutorial](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/net_tutorial.md).
 That would have saved me quite a bunch of pain =.=
Thank god for the #raspberrypi IRC channel on freenode clearing stuff up. I'm off to dismounting all of my PI's and soldering.

closed 2019-01-03 by ghollingworth

JamesH65, 2019-01-03:
@SairesArt If you can provide a PR for the documentation change required, I'd be more than happy to review it and add to the repo as necessary. I'm entirley unfamiliar with this area, so probably better coming from someone with the experience of knowing where the docs are lacking!

https://www.raspberrypi.org/documentation/using-github.md

ghollingworth, 2019-01-03:
Just done

JamesH65, 2019-01-03:
Thanks, see https://github.com/raspberrypi/documentation/pull/1056.