raspberrypi/firmware #1082: Raspberry PI 3A+ support

Romaxx, 2018-12-14:
Hi,
3A+ board use the same broadcom chip as 3B+, but is the bcm2710-rpi-3-b-plus.dtb compatible with 3A+? 
I've made a custom Yocto build (64bits) based on meta-raspberrypi (using broadcom/*.dtb), on 3A+, xrandr show HDMI-1 but dont detect any display, perhaps the initialization of LAN7515 (not implemented on 3A+) throw an error and break something ?

JamesH65, 2019-01-08:
@Romaxx OK to close this? Question is answered,and the rest of the problems, if you havent fixed them already, are better handled on the forum.

JamesH65, 2019-01-08:
Prety much the only difference. But this conversation is better handled on the forum - lots more people on there doing 64bit stuff who will never see this issue on here.

Romaxx, 2019-01-09:
well, same issue with 32bit version.. with the same SD card, 3B and 3B+ works, 3A+ freeze.
3B+ have Soc "BCM2837B01FSBG TA1824 P20 058-01 N2 W" and 3A+ "BCM2837B01FSBG TA1829 P20 084-01 N2 W". where is the forum ? @JamesH65

JamesH65, 2019-01-09:
https://www.raspberrypi.org/forums/

As for it still failing on a 32bit build, there must be something wrong with your kernel/Yocto setup.  As to what that could be, no idea. Have you cross referenced your kernel setting with the ones in our kernel?

JamesH65, 2019-01-15:
@Romaxx Can this be closed?

Romaxx, 2019-01-17:
@JamesH65 Yes

closed 2019-01-17 by pelwell

bvarner, 2019-11-12:
@Romaxx and @JamesH65 I've been having this same issue using yocto to build images for the last few months. Eventually I figured out if I disable vc4graphics, the machine will boot as expected and detect the HDMI display.

I found this issue while searching for things.

From the messages what I've gathered is that:
It looked like there were device tree changes that needed to be set for the 3A+.
Then the forums get mentioned.
Then the issue gets closed with no apparent update?

Was there something that came from this?

pelwell, 2019-11-12:
The 3B+ dtb is fine on a 3A+. Only Yocto users seem to be having problems.

You could try adding an explicit `device_tree=bcm2710-rpi-3-b-plus.dtb` to config.txt,b and/or using `sudo vcdbg log msg` to read the boot log from the firmware.