raspberrypi/firmware #1083: rpi3b core timer in aarch64 mode

mlelstv, 2018-12-17:
I am running a NetBSD kernel for aarch64 on a RPI3b. When busy, after some time, the timer interrupt stops. The timer interrupt is driven by the virtual ARMv8 generic timer. I have tracked this to the
core timer that apparently is reset to a low value, possibly zero. It then takes a long time to reach
the comparison value and trigger the next timer interrupt. Writing a reasonable value to the core
timer registers makes the interrupt trigger again and the clock continues.

Since nothing in the kernel would regularly write to the core timer and cause such a reset I was
wondering if the firmware can do this under some condition?

pelwell, 2018-12-17:
The firmware/VPU does not have access to the per-core timers, so it can't be responsible for the resets.

mlelstv, 2018-12-17:
The per-core timers are driven by the 'core timer', a local peripheral at 0x400001c/020. Is that also unreachable by the VPU ?

popcornmix, 2018-12-17:
GPU cannot access peripherals at 0x4xxxxxxx addresses (it see's it's own L2 cache at that address).

mlelstv, 2018-12-17:
Understood. Thanks for the answer.

closed 2018-12-17 by mlelstv