raspberrypi/firmware #1084: ISP BGR24 to I420 conversion problem after firmware update

luiscgalo, 2018-12-22:
Hello RPi support,

I'm using a custom application developed by me which performs conversion of BGR24 images to I420.
My application was working fine during the last months.
However, yesterday, after updating Raspberry Pi firmware to its latest version, the color conversion no longer works as expected.
When the problem appears, the colors on the I420 output are not correct (it is something strange that increases brightness and corrupts colors of the image).

From my analysis, the commit where the problem starts appearing:
    Nov 21, 2018 - 7dcd64144a10f5d6f81513ae36df93bb9b70f3fc (4.14.82) -> Still works fine
    Nov 29, 2018 - e5585fc7a074ab6614cff8e41cf0fa041851fd7f (4.14.84) -> Problem starts appearing

I've created a very simple demo application that reads a BGR24 binary file, converting it to I420 using ISP component so you can try to replicate the issue I'm currently facing.
I've also added "correct" and "wrong" image files so you can see what's happening (I visualize the raw BGR24 / I420 files at http://rawpixels.net/).

The demo app and test images are available on the following link: [isp-issue.zip](https://github.com/raspberrypi/firmware/files/2705549/isp-issue.zip)

Is there any relevant change on the ISP component after referred commit?
Since my application was working fine during last months, I'm quite surprise why it does not work anymore after upgrading to latest firmware versions...

Thanks in advance for your help.

Regards,
Luís Galo

pelwell, 2018-12-22:
There were quite a few potential suspects between those two releases:
```
firmware: MMAL: Include OPAQUE in the list of supported encodings where appropriate
See: https://www.raspberrypi.org/forums/viewtopic.php?f=67&t=226416

firmware: video_render: asserting on the buffer being unlocked is invalid
firmware: video_render: Further fix to a vcos_verify

firmware: IL hvs: Fail component create if the HVS or TXP interrupts are masked
firmware: Clean up masked interrupt handling - moves it to intctrl
firmware: vcinclude: Clean up INTERRUPT_xxx defines to use one place
firmware: vcfw: camera_subsystem: Stop cameras being detected if ARM has Unicam instance

firmware: vc_image: Fix up vc_image_bits_per_pixel for YUYV variants and BGR888
firmware: vc_image: Make the alternate path in yuv420 to rgb888 support bgr888
firmware: vc_image: Add an rgb_stripe_swap_ext which takes pitches
firmware: vc_image_convert: yuv420 to bgr888 failed on (width&31) != 0

firmware: IL isp: Use gamma block to shift 10bpp YUV to the correct pipe depth
firmware: video_encode: Always set the input port nBufferSize to match the image
```

6by9, 2018-12-23:
Curious. 
```firmware: IL isp: Use gamma block to shift 10bpp YUV to the correct pipe depth``` is the only commit that is likely to have had any effect on that, but the gamma block shouldn't have been enabled for 24bpp BGR to I420 (it's only used for 10bpp YUV input).

I don't have any kit set up at the moment due to the holidays, but will have a look when I get a chance.

luiscgalo, 2018-12-23:
Hi pelwell and 6by9.
Thanks for your feedback.
Yes, it is strange but indeed something is different after Nov 29, 2018 commits.

But we can try to solve this after holidays. :)

Thanks again for your help.
A merry Christmas and happy new year ;)

luiscgalo, 2019-01-08:
Ok,
Indeed, something has changed on the firmware which explains the problem on the colour conversion from BGR to I420.
I will wait for news and future update as soon as the problem is solved.
Until that, I'l keep using the older firmware version where conversion works properly.

Thanks for your support 6by9.

popcornmix, 2019-01-09:
Now in latest rpi-update firmware

luiscgalo, 2019-01-15:
Hi all,
Thanks for your effort and for the update on RPi firmware.
I can confirm that, with the latest version of firmware, everything works perfectly.

Issue considered as closed.

closed 2019-01-15 by luiscgalo