raspberrypi/firmware #1085: Running omxplayer together with an OpenGL surface on dispmanx results in a black screen when --blank is passed

carlonluca, 2018-12-24:
Hello,
this seems like a very stupid issue, but still I'm not finding a possible explanation. I'm trying to use omxplayer to render a video while my application runs. My application uses Qt, which means it creates a dispmanx layer and renders with OpenGL into it.
What I see is that everything works properly (as far as I can see) until I add the --blank option to the omxplayer instance with a color not entirely transparent. If I add, for instance, --blank=0xFFFFFF00, the screen turns entirely black and, at a certain point, the TV reports no signal from the raspberry. Killing omxplayer brings back the signal and everything is shown properly. If I run, in the same situation, omxplayer with --blank=0x00FFFF00, everything seems fine.

I was able to simplify and replicate the issue by simply running a dispmanx layer and running omxplayer above. I used this code to create the dispmanx layer:

```
#include <stdio.h>
#include <stdlib.h>
#include <stdarg.h>
#include <assert.h>
#include <unistd.h>
#include <sys/time.h>
#include "bcm_host.h"
#define WIDTH   1000
#define HEIGHT  1000
#ifndef ALIGN_UP
#define ALIGN_UP(x,y)  ((x + (y)-1) & ~((y)-1))
#endif
typedef struct
{
    DISPMANX_DISPLAY_HANDLE_T   display;
    DISPMANX_MODEINFO_T         info;
    void                       *image;
    DISPMANX_UPDATE_HANDLE_T    update;
    DISPMANX_RESOURCE_HANDLE_T  resource;
    DISPMANX_ELEMENT_HANDLE_T   element;
    uint32_t                    vc_image_ptr;
} RECT_VARS_T;
static RECT_VARS_T  gRectVars;
static void FillRect( VC_IMAGE_TYPE_T type, void *image, int pitch, int aligned_height, int x, int y, int w, int h, int val )
{
    int         row;
    int         col;
    uint16_t *line = (uint16_t *)image + y * (pitch>>1) + x;
    for ( row = 0; row < h; row++ )
    {
        for ( col = 0; col < w; col++ )
        {
            line[col] = val;
        }
        line += (pitch>>1);
    }
}

int main(void)
{
    RECT_VARS_T    *vars;
    uint32_t        screen = 0;
    int             ret;
    VC_RECT_T       src_rect;
    VC_RECT_T       dst_rect;
    VC_IMAGE_TYPE_T type = VC_IMAGE_RGB565;
    int width = WIDTH, height = HEIGHT;
    int pitch = ALIGN_UP(width*2, 32);
    int aligned_height = ALIGN_UP(height, 16);
    VC_DISPMANX_ALPHA_T alpha = { DISPMANX_FLAGS_ALPHA_FROM_SOURCE | DISPMANX_FLAGS_ALPHA_FIXED_ALL_PIXELS, 
                             120, /*alpha 0->255*/
                             0 };
    vars = &gRectVars;
    bcm_host_init();
    printf("Open display[%i]...\n", screen );
    vars->display = vc_dispmanx_display_open( screen );
    ret = vc_dispmanx_display_get_info( vars->display, &vars->info);
    assert(ret == 0);
    printf( "Display is %d x %d\n", vars->info.width, vars->info.height );
    vars->image = calloc( 1, pitch * height );
    assert(vars->image);
    FillRect( type, vars->image, pitch, aligned_height,  0,  0, width,      height,      0xFFFF );
    FillRect( type, vars->image, pitch, aligned_height,  0,  0, width,      height,      0xF800 );
    FillRect( type, vars->image, pitch, aligned_height, 20, 20, width - 40, height - 40, 0x07E0 );
    FillRect( type, vars->image, pitch, aligned_height, 40, 40, width - 80, height - 80, 0x001F );
    vars->resource = vc_dispmanx_resource_create( type,
                                                  width,
                                                  height,
                                                  &vars->vc_image_ptr );
    assert( vars->resource );
    vc_dispmanx_rect_set( &dst_rect, 0, 0, width, height);
    ret = vc_dispmanx_resource_write_data(  vars->resource,
                                            type,
                                            pitch,
                                            vars->image,
                                            &dst_rect );
    assert( ret == 0 );
    vars->update = vc_dispmanx_update_start( 10 );
    assert( vars->update );
    vc_dispmanx_rect_set( &src_rect, 0, 0, width << 16, height << 16 );
    vc_dispmanx_rect_set( &dst_rect, ( vars->info.width - width ) / 2,
                                     ( vars->info.height - height ) / 2,
                                     width,
                                     height );
    vars->element = vc_dispmanx_element_add(    vars->update,
                                                vars->display,
                                                0,               // layer
                                                &dst_rect,
                                                vars->resource,
                                                &src_rect,
                                                DISPMANX_PROTECTION_NONE,
                                                &alpha,
                                                NULL,             // clamp
                                                VC_IMAGE_ROT0 );
    ret = vc_dispmanx_update_submit_sync( vars->update );
    assert( ret == 0 );
    while (1) sleep( 10 );
    vars->update = vc_dispmanx_update_start( 10 );
    assert( vars->update );
    ret = vc_dispmanx_element_remove( vars->update, vars->element );
    assert( ret == 0 );
    ret = vc_dispmanx_update_submit_sync( vars->update );
    assert( ret == 0 );
    ret = vc_dispmanx_resource_delete( vars->resource );
    assert( ret == 0 );
    ret = vc_dispmanx_display_close( vars->display );
    assert( ret == 0 );
    return 0;
}
```

If I let the previous code running, and run above:

`omxplayer BigBuckBunny1080p.mp4 --blank=0x00FFFF00 --aspect-mode letterbox --win 0,0,1000,1000 --layer 2`

everything works as I'd expect. If I run this instead:

`omxplayer BigBuckBunny1080p.mp4 --blank=0xFFFFFF00 --aspect-mode letterbox --win 0,0,1000,1000 --layer 2`

I get the proper behavior for a second or so, then black screen, and after a while no more signal. Killing omxplayer brings the signal back.

Any idea why this happens?
Thanks!

popcornmix, 2018-12-24:
Yes, it's too many layers for the HVS (hardware video scaler) to compose in real time.
Run ``vcgencmd dispmanx_list`` (e.g. though ssh) when omxplayer is running to see exactly what layers are present.

``dispmanx_offline=1`` in config.txt will likely this this issue by avoiding real-time composition and composing to an off-screen buffer. But that is more expensive and will harm performance.

Best options is to use a newer omxplayer which can provide the same behaviour of --blank without an extra HVS layer. See: https://github.com/popcornmix/omxplayer/commit/a70e9b4b38e9568eac5e7ec8d19b5b4b70165bf5
Note: latest omxplayer requires a recent firmware version.

carlonluca, 2018-12-26:
Hello @popcornmix, thank you for your answer.
Yes, if possible I'd prefer to avoid harming performance.
I updated the system and the firmware (via rpi-update):

```
$ vcgencmd version
Dec 17 2018 23:56:39 
Copyright (c) 2012 Broadcom
version da468960fe03ecbaa8e3f1ee01c7217c3bd01fa8 (clean) (release)
```

and I built omxplayer from git:

```
$ ./omxplayer -v
omxplayer - Commandline multimedia player for the Raspberry Pi
        Build date: Wed, 26 Dec 2018 01:09:18 +0000
        Version   : 5da1379 [master]
        Repository: https://github.com/popcornmix/omxplayer.git
```

I ran the same test described above, but I'm getting the same result. I tried to list the layers as you suggested:

```
$ vcgencmd dispmanx_list
display:2 format:XRGB8888 transform:0 layer:-127 src:0,0,1824,984 dst:48,48,1824,984 cost:1103 lbm:0 
display:2 format:RGB565 transform:0 layer:0 src:0,0,1000,1000 dst:460,40,1000,1000 cost:506 lbm:0 
display:2 format:ARGB8888 transform:100000 layer:1 src:0,0,1,1 dst:0,0,0,0 cost:1186 lbm:1024 
display:2 format:YUV_UV transform:0 layer:2 src:0,0,1920,1080 dst:0,219,1000,562 cost:2283 lbm:16384
```

Did I need to pass a different param instead of --blank?
Another question: I see 4 layers here. Is 3 the max number of layers I can have?
Thank you!

popcornmix, 2018-12-27:
Why are you using ``--blank`` ?
Latest omxplayer should automatically blank the black bars when video is playing fullscreen.
When video is playing windowed I would assume you didn't want the remaining area to be blanked?

The max number of layers depends on a few factors. RGB8888 requires twice the memory bandwidth of RGB565. Scaled images (esp vertically) cost more than unscaled. You can have more layers if they don't overlap vertically (e.g. a layer in top half and a layer in bottom half of screen is cheaper than overlapping).
Overclocking can allow more complexity.

Two layers should be fine. Three depends on complexity. Four usually won't work (when mostly full screen).

From your list:
The first layer is probably the linux console (which is probably not being used and could be removed to improve performance).
The second is an RGB565 1000x1000 layer, I guess from your application
The third is the ``--blank`` layer
The fourth is the video layer.

carlonluca, 2018-12-27:
I see, thank you for the explanation. So the bars can only be black when in fullscreen and transparent when windowed when --blank is not passed?

popcornmix, 2018-12-27:
Yes, windowed mode behaviour hasn't changed. It is assumed you want video as a smaller window over current display.
For fullscreen, seeing the console in the bars is usually undesirable so default behaviour is to set a flag that ignores layers below it. This is cheaper than the old ``--blank`` option which renders lower layers and then renders a blank layer on top so they are hidden.

In latest omxplayer ``--blank`` with default opaque black setting can be skipped and the cheaper ignore lower layers flag is used. If you want non-opaque settings, then that isn't possible and the more expensive solution is used.

carlonluca, 2019-01-02:
Thank you for the information.
You said it could be possible to remove the layer used by the console. How can it be done?
As this is the expected behavior I guess this issue can be closed. Thank you.

popcornmix, 2019-01-02:
If you run omxplarer with -r it has the side effect of removing existing dispman layers (including console). If you check /usr/bin/omxplayer you can see the logic that restores them.

Also ``fbset -xres 16 -vxres 16 -yres 16 -vyres 16`` will make the framebuffer small enough to have little cost.

JamesH65, 2019-01-09:
As OP suggested this can be closed....

closed 2019-01-09 by JamesH65