raspberrypi/firmware #1088: RPI B3+ gets stuck on reboot when doing Netboot

Flole998, 2019-01-04:
As mentioned at https://github.com/raspberrypi/firmware/issues/1078#issuecomment-451233345 in #1078 there is an issue when the pi reboots and gets stuck somewhere, apparently the issue is already reproducable.

ghollingworth, 2019-01-06:
Yes, I've got a simple reproduction, which is to use 

`echo b > /proc/sysrq-trigger`
Or Alt + PrtScr + b on the keyboard

Which causes the lock up on reboot...  Currently it is locking up (and crashing hard) while trying to read from one of the files, this is caused by an exception or interrupt occuring during the bootcode.  Unclear at the moment what the cause is.

popcornmix, 2019-01-09:
Potential fix now in latest rpi-update firmware

Flole998, 2019-01-09:
> Potential fix now in latest rpi-update firmware

Are you sure? Note that this is not the netboot-on-different Subnet issue from #1078

pelwell, 2019-01-10:
Yes - read the commit message:
```
...
bootcode: Reset WiFi and BT devices before resetting the expander
See: raspberrypi/firmware#1088
```

Flole998, 2019-01-11:
Seems indeed fixed! Great!

closed 2019-01-11 by Flole998