raspberrypi/firmware #1089: mmal_vc_diag mmal-stats, what do these values mean?

zoff99, 2019-01-08:
I am using a rpi3B+ with latest firmware updated with rpi-update
```
vcgencmd version
Dec 18 2018 00:00:23 
Copyright (c) 2012 Broadcom
version da468960fe03ecbaa8e3f1ee01c7217c3bd01fa8 (clean) (release)

cat /boot/.firmware_revision
da5948d8ef8354557732d9c8f5ad5e7e24374a69
```

for some reason these 2 values are not the same.
now I am encoding and decoing 25fps H264 frames with HW acceleration.

mmal-stats gives me this strange output:

```
sudo /opt/vc/bin/mmal_vc_diag mmal-stats
component		port		buffers		fps	delay
ril.video_decode    	0 [in ]rx	9351      	15.4	859466
ril.video_decode    	0 [in ]tx	9351      	15.4	859367
ril.video_decode    	0 [out]rx	9369      	15.5	861336
ril.video_decode    	0 [out]tx	9362      	15.5	866190
ril.video_encode    	0 [in ]rx	15111     	25.0	52717
ril.video_encode    	0 [in ]tx	15111     	25.0	52705
ril.video_encode    	0 [out]rx	15234     	25.2	69656
ril.video_encode    	0 [out]tx	15232     	25.2	66889
ril.image_encode    	0 [in ]rx	0         	 0.0	0
ril.image_encode    	0 [in ]tx	0         	 0.0	0
ril.image_encode    	0 [out]rx	0         	 0.0	0
ril.image_encode    	0 [out]tx	0         	 0.0	0
ril.video_render    	0 [in ]rx	0         	 0.0	0
ril.video_render    	0 [in ]tx	0         	 0.0	0
ril.camera          	0 [out]rx	0         	 0.0	0
ril.camera          	0 [out]tx	0         	 0.0	0
ril.camera          	1 [out]rx	15137     	 9.0	957929335
ril.camera          	1 [out]tx	15127     	 9.0	957922659
ril.camera          	2 [out]rx	0         	 0.0	0
ril.camera          	2 [out]tx	0         	 0.0	0
```

what does this mean? what is delay? is it im ms or us?
why does it show 9.0fps and 15.5fps and 25.0fps?
i am confused, is there a document to explain that in detail?

thank you.

JamesH65, 2019-01-09:
I don;t think anyone here really knows much about this diagnostic tool, but I will ask around.  It's quite possible it has bugs in it. Taking a quick look at the code, the delay looks to be calculated correctly, its just the maximum delay found between buffers received.  One question, are you changing the clock time on the Pi at any point? Because if the clock on the Pi changes in any way, this could affect that calculation (this is a known issue with the ARM side timer that we are looking in to - its been there for years though)

JamesH65, 2019-01-09:
Ah, found a bug! Fixed, and will give it a quick test. Will post the executable once ready so you can test it.

6by9, 2019-01-09:
mmal_vc_diag was a dev tool from back in the Broadcom days. It's not one I use routinely, and I'm not sure how useful the stats are in general cases, more in profiling specific pipelines.

The stats are cumulative from the point the component is created. In the case of the V4L2 driver this is when it is loaded. If you run start capturing, stop, and then resume, I would expect the numbers to give odd results. It will also mean that the biggest delay is that stopped period.
You can use ```mmal-vc-diag mmal-stats reset``` to reset them.

A further quirk is that H264 header bytes come from the encoder with a timestamp of 0, therefore you are likely to get some odd results with H264.

NB Full source for it at https://github.com/raspberrypi/userland/blob/master/interface/mmal/vc/mmal_vc_diag.c
It's dumping out the stats from each port as setup by the MMAL core. https://github.com/raspberrypi/userland/blob/master/interface/mmal/core/mmal_port.c#L96
https://github.com/raspberrypi/userland/blob/master/interface/mmal/mmal_common.h#L64
https://github.com/raspberrypi/userland/blob/master/interface/mmal/core/mmal_port.c#L1417

zoff99, 2019-01-09:
thanks guys.
@6by9 what tool do you use to get useful info?
we are trying to squeeze a lot out of the Pi's GPU. 720p@25fps full duplex with end-to-end encryption, plus some other stuff.

zoff99, 2019-01-09:
we have tryied all sorts of combinations, most do not work properly together.
now we do h264_mmal decoding (with ffmpeg) , h264 encoding via 4vl2, and displaying the video with OMX.
thats so far the best combination.

zoff99, 2019-01-09:
still leaves the 9.0fps when the camera is actually giving out 25fps
```
ril.camera          	1 [out]rx	15137     	 9.0	957929335
```
strange this

6by9, 2019-01-09:
It depends where the bottle-neck is.

```sudo vcdbg hist gnuplot``` gives you the task history of the VPUs for the last N task switches. Display with ```gnuplot task.gpt```. If that's maxed out then you're probably stuck.

```htop``` obviously tells you the ARM loading.

There are stats gathered by the AXI bus arbiters to give an idea over sdram bandwidth usage. ```dtparam=axiperf=on``` to enable them, and then it creates directories in /sys/kernel/debug/raspberrypi_axi_monitor. Interpreting the numbers is a bit of an art as only the VPU can see some of the arbiters, whilst the ARM can see the others. They can also only monitor a couple of sources at a time.
Ask JamesH65 as he wrote it!

Memory bandwidth is likely to be your issue, but you don't give enough details about the pipelines. Using ffmpeg and IL will most likely be doing a memcpy of each and every image frame from ARM memory to GPU memory or vice versa. At 720P I420 that's 1.3MB/frame, so multiply it up by the frame rate and trips back and forth and it starts hurting quite rapidly.
If you either tunnel with IL, use MMAL zero copy, or mmal_connections with the ```MMAL_CONNECTION_FLAG_TUNNELLING``` flag set, then you can avoid those copies. 

h264 encoding from the bcm2835-v4l2 driver? I query as I'm working on a V4L2 memory to memory H264 encoder at the moment, and there are definite gremlins still lurking in there. (It's on the rpi-4.19.y branch only)

zoff99, 2019-01-13:
thanks for all the info. i think it's working ok now

closed 2019-01-13 by zoff99