raspberrypi/firmware #1090: Possibility to modify HDMI timings on the fly VIA C

alphanu1, 2019-01-13:
Hi,

Is it a all possible to modify the hmdi_timming from my program using C? 

I am using stretch light so there is no desktop environment. When I attempt to use system(), exec() or popen() calls with vcgencmd my application looses focus and I am stuck back at terminal. From this point I am unable to get my application back. 

Sorry if this has been asked and answered before. I have been unable to find any information on it.

Regards,

Ben

6by9, 2019-01-14:
vcgencmd is just C, calling in to VCHIQ to request things from the firmware. source at https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/gencmd/gencmd.c

https://www.raspberrypi.org/forums/viewtopic.php?t=229203#p1406112 gives an example of using the same calls yourself. That example is tweaking the scaling parameters, but the same approach should work for hdmi_timings.

6by9, 2019-01-16:
What exactly is your application doing? Writing to the frame buffer, or using DispmanX or other GPU API to render stuff?

When you say terminal, do you mean the fullscreen console, or do you mean a terminal window within X11?

Resetting the HDMI timings is likely to have to reset the whole display pipeline as it potentially changes the screen resolution and/or geometry. It's effectively the same as using ```tvservice -e``` to switch standard HDMI mode, and that removes everything including the framebuffer.

alphanu1, 2019-01-17:
It's writing to the frame buffer. I had to reinit the video driver to pull it back to the buffer. All is good now.

Cheers for helping @6by6

alphanu1, 2019-01-17:
Yeah I'm happy. Sorry about the wrong tags @6by9

closed 2019-01-17 by alphanu1