raspberrypi/firmware #1092: Raspberry Pi 3 bluetooth connect and disconnect

ram94638, 2019-01-17:
I have RPI 3B, When I connect to device with using password It connect for few seconds then disconnect, After that I installed blueman and try to connect then it configured serial port and get connected. But I turn off to bluetooth,pi or device its not connect automatically. Again same process by manually connect. Although I have paired and trust to that device. Any solution?

ram94638, 2019-01-17:
Already Done these changes

Edit /etc/systemd/system/dbus-org.bluez.service

sudo nano /etc/systemd/system/dbus-org.bluez.service

Add a -C to the line
ExecStart=/usr/lib/bluetooth/bluetoothd
So that it looks like
ExecStart=/usr/lib/bluetooth/bluetoothd -C

And then add the following line below it
ExecStartPost=/usr/bin/sdptool add SP

So now the file has
ExecStart=/usr/lib/bluetooth/bluetoothd -C
ExecStartPost=/usr/bin/sdptool add SP

pelwell, 2019-01-17:
1. Which operating system (and which version/release) are you running?
2. What device is it?
3. Have you tried replacing the batteries in the device?

If you are running Raspbian then installing Blueman was not a good idea - it is likely to interfere with the built-in Raspbian BT support.
If you are not running Raspbian, I am unlikely to be able to support you. Try installing a Raspbian image onto a spare SD card and see if the connection still drops after pairing.

ram94638, 2019-01-17:
I am using Raspbian Strech(9)

ram94638, 2019-01-18:
?

pelwell, 2019-01-18:
You answered one out of three questions - your move.

ram94638, 2019-01-18:
2.That is customized device with embedded bluetooth v3,class 1
3. Battery replaced and it also working on DC power

pelwell, 2019-01-18:
Starting with a clean Raspbian, configure it as before but without installing BlueMan. Then before pairing your device, run the following command in a terminal window:
```
$ sudo btmon -w pair.bts
```
It will display messages to the window. Now pair and connect your device, waiting for it to disconnect. You could even connect it a second time for good measure.

When you think you have seen the failure a few times, kill the btmon capture with Ctrl-C, then upload pair.bts somewhere I can download it.

N.B. There is only so far I can go to diagnose a fault connecting to a custom device, but the capture might reveal if it is something simple.

jhdgghost, 2019-01-19:
Just tested in my rpi3b+ with librelelec 9 build 2019-#0119. The a2dp audio disconnect after 5-15 minutes working with pulse audio.

pelwell, 2019-01-21:
[ @jhdgghost This is a different issue to the one you are experiencing. ]

1. The log shows an initial pairing request using the default PIN of 0000, which is rejected
2. This is followed by a second attempt with a real PIN after a delay (presumably while you were typing it in), which is accepted. At this point, pairing has been successful.
3. The Pi then sends an L2CAP "Extended features supported" request, which is "Not supported".
4. An L2CAP Connection Request comes next, which is successful.
5. Both ends agree that the MTU is 256.
6. An "SDP: Service Search Attribute Request" for L2CAP gets a long response:
```
> ACL Data RX: Handle 12 flags 0x02 dlen 147                   [hci0] 88.459415
      Channel: 64 len 143 [PSM 1 mode 0] {chan 0}
      SDP: Service Search Attribute Response (0x07) tid 0 len 138
        Attribute bytes: 135
          Attribute list: [len 54] {position 0}
            Attribute: Service Record Handle (0x0000) [len 2]
              0x00010000
            Attribute: Service Class ID List (0x0001) [len 2]
              UUID (3) with 2 bytes [0 extra bits] len 3
                Serial Port (0x1101)
            Attribute: Protocol Descriptor List (0x0004) [len 2]
              Sequence (6) with 3 bytes [8 extra bits] len 5
                UUID (3) with 2 bytes [0 extra bits] len 3
                  L2CAP (0x0100)
              Sequence (6) with 5 bytes [8 extra bits] len 7
                UUID (3) with 2 bytes [0 extra bits] len 3
                  RFCOMM (0x0003)
                Unsigned Integer (1) with 1 byte [0 extra bits] len 2
                  0x01
            Attribute: Browse Group List (0x0005) [len 2]
              UUID (3) with 2 bytes [0 extra bits] len 3
                Public Browse Root (0x1002)
            Attribute: Unknown (0x0100) [len 2]
              AMP-SPP [len 8]
          Attribute list: [len 73] {position 1}
            Attribute: Service Record Handle (0x0000) [len 2]
              0x00010001
            Attribute: Service Class ID List (0x0001) [len 2]
              UUID (3) with 16 bytes [0 extra bits] len 17
                00000000-deca-fade-deca-deafdecacaff
            Attribute: Protocol Descriptor List (0x0004) [len 2]
              Sequence (6) with 3 bytes [8 extra bits] len 5
                UUID (3) with 2 bytes [0 extra bits] len 3
                  L2CAP (0x0100)
              Sequence (6) with 5 bytes [8 extra bits] len 7
                UUID (3) with 2 bytes [0 extra bits] len 3
                  RFCOMM (0x0003)
                Unsigned Integer (1) with 1 byte [0 extra bits] len 2
                  0x02
              Sequence (6) with 3 bytes [8 extra bits] len 5
                UUID (3) with 2 bytes [0 extra bits] len 3
                  PnP Information (0x1200)
            Attribute: Browse Group List (0x0005) [len 2]
              UUID (3) with 2 bytes [0 extra bits] len 3
                Public Browse Root (0x1002)
            Attribute: Unknown (0x0100) [len 2]
              AMP-iAP [len 8]
        Continuation state: 0
```
This suggests the device is an "AMP-SPP" and an "AMP-iAP".

7. A second "SDP: Service Search Attribute Request", this time for PnP Information, gets just the "AMP-iAP" response.
8. 2 seconds later, the Pi sends an L2CAP Disconnection Request.
9. A further ~20 seconds later, the device disconnects.

Perhaps the disconnection is because the device is expecting to be used in a particular way, and when the Pi doesn't do it for 20 seconds it gives up.

ram94638, 2019-01-22:
This issue is not only with my particular device, same issue on other bluetooth device.

kilian579, 2019-03-21:
I'm facing a similar issue using a bluetooth button (extended keyboard). The time it actually stays connected seems to range from 5 to 20 minutes. After the connection dropped, it can be regained by running `bluetoothctl` in the terminal. This step requires closing other programs running in the shell.

chuwwh, 2019-06-09:
I am using "HP Bluetooth Keyboard T1000" for months smoothly. Everyday, it runs thru many cycles of connect, sleep(battery save), disconnect, wake up, connect,  and the Raspberry Pi 3 B+ handle it smoothly.

But since last week I did a "apt-get update" and "apt-get upgrade", I found my keyboard disconnected.
Pairing is successful and keyboard can work as long as it does not go into sleep and disconnect. Once it disconnected, no way it can reconnect. The Pi just repeatedly  "connected: yes" and immediately "connected: no".

I trouble shoot it for days in different ways and now notice that when the keyboard got paired under the upgraded environment, it has a class 0x002540.  But instead of keep using 0x002540, the Pi recognizes it as 0x000540 during reconnect which is then followed by disconnect.   

Now I have to fallback my Pi OS, which pair my keyboard with class 0x000540.

pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian

[HP Bluetooth Keyboard T1000]# info
Device 20:73:24:03:36:97
	Name: HP Bluetooth Keyboard T1000
	Alias: HP Bluetooth Keyboard T1000
	Class: 0x000540
	Icon: input-keyboard

thomaslc66, 2019-07-23:
Facing same issue with an galaxy s5 device on android.

mudassarn, 2019-12-31:
Facing the same issue with multiple bluetooth speakers. It connects for 15-20 seconds and then disconnect afterwards. Is there a solution?

rob-meiner, 2020-08-04:
Did this ever get resolved?  I am having the same problem with a Raspberry Pi 4b. 

This happens with every bluetooth device that I try.  

```
pi@antonio:~ $ sudo systemctl status hciuart.service
● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/lib/systemd/system/hciuart.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2020-08-04 14:02:57 CEST; 31min ago

Aug 04 14:02:27 antonio systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Aug 04 14:02:57 antonio btuart[5279]: Initialization timed out.
Aug 04 14:02:57 antonio btuart[5279]: bcm43xx_init
Aug 04 14:02:57 antonio systemd[1]: hciuart.service: Control process exited, code=exited, status=1/FAILURE
Aug 04 14:02:57 antonio systemd[1]: hciuart.service: Failed with result 'exit-code'.
Aug 04 14:02:57 antonio systemd[1]: Failed to start Configure Bluetooth Modems connected by UART.
```

I updated to latest rpi-config using ```sudo rpi-update``` and rebooted.

Bluetoot still disconnects within 2-5 seconds of connecting (I xx'ed out the MAC addresses)

```
[CHG] Device 44:78:XX:XX:XX:XX Connected: yes
Request confirmation
[NEW] Device 45:A0:XX:XX:XX:XX 45-A0-XX-XX-XX-XX
[NEW] Device 60:6D:XX:XX:XX:XX KDL-55W807C
yes
[CHG] Device 44:78:XX:XX:XX:XX Connected: no
```
Let me know if someone has any ideas.
Thanks!
Mark