raspberrypi/firmware #1094: [Question] Multiple DSI touchscreens

jriwanek, 2019-01-18:
So, it's a known fact that the Compute module has two DSI ports, but it's as yet unknown if the firmware currently (or will) support multiple DSI screens at once, I'd like to know if it's possible, or could be added to a future firmware.

JamesH65, 2019-01-18:
That might work with the multi framebuffer test code, but I don't think I have ever tried it. 

See https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=216399

jriwanek, 2019-01-18:
Looks like me or someone else needs to get a second DSI display. xD
Thanks for the link!

JamesH65, 2019-01-18:
Cannot garantee it will work, I've don't remember trying it....

jriwanek, 2019-01-18:
You said in the thread that you didn't, so I'd assume you didn't, if possible, it'd be really great if you could as I currently only have the one DSI display.

It's also mentioned that you can't use DSI at the same time as DPI on the thread, is that true or no?

JamesH65, 2019-01-18:
Correct that you cannot use DPI and DSI at the same time -  I did try that.

jriwanek, 2019-01-18:
Is that a hardware limitation, or something which may change in the future?

JamesH65, 2019-01-18:
HW limitation. Cannot remember the exact reason.

jriwanek, 2019-01-18:
I'd assume that'd apply to HDMI + DSI + DPI too, which leaves:
HDMI + DSI
HDMI + DPI
DSI + DSI
HDMI + DSI + DSI

as possible candidates.

JamesH65, 2019-01-18:
HDMI+DSI and HDMI+DPI both work, not tested the others. As it stands the firmware is limited to two displays, so the last won't work, however, the code can be easily modified to go to three, I'm just not sure if the HW could support it, or even keep up - three displays would probably overload the HVS, especially if there are any transposes going on. That's a LOT of data to be throwing around.

jriwanek, 2019-01-18:
Ofcourse; I'm planning to buy or obtain a second touchscreen in a month or so, so if you could provide an experimental firmware which supports all three, I'd be happy to test.

jriwanek, 2019-01-18:
Darn, I just remembered the Composite output too!

6by9, 2019-01-18:
The firmware driver and configuration for the display doesn't allow for two displays. The driver uses a single static structure for the state, and the configuration in dt-blob.bin doesn't allow for 2 displays (compare to the camera config which does).

Eric's driver (vc4-drm-kms) may allow for both as both DSI devices are defined in device tree. Whether it works with both or not, and what limitations there are on how many devices can be used simultaneously I don't know for certain.
https://github.com/anholt/linux/wiki/Raspberry-Pi-and-other-MIPI-DSI-display-panels

jriwanek, 2019-01-18:
Sounds like it could be made possible with some firmware changes, the question is whether that'd be something on the cards in the foundation's eyes.

I maintain my request for a firmware supporting three displays for the purpose of trying with HDMI + DSI/DPI + Composite.

JamesH65, 2019-01-18:
I'm dubious as to 3 screen working, and getting two DSI, as 6by9 says (just checked the code), would require changes to the LCD driver AND the touchscreen driver to cope with multiple instances. Not impossible, but not high priority as it's a very small use case and quite a lot of work.  Your request is noted however!

6by9, 2019-01-18:
Reading the spec sheet it says:

> There are three pixel valves.  Pixelvalve0 can be used by DSI0 (which is 2 data lane DSI) or DPI. Pixelvalve1 can be used by DSI1 (which is 4 data lane DSI) or SMI. Pixelvalve2 can be used by HDMI (which is 4 data lane DSI) or SDTV.

So it should be possible to run:
- DSI0+DSI1+HDMI
- DPI+DSI1+HDMI
- DSI0+DSI1+SDTV
- DPI+DSI1+SDTV

(Your requested combination of HDMI + DSI/DPI + Composite SDTV is not possible).

Enforcing those limitations via the configuration is going to be nasty, but leaving it as caveat emptor should you be messing with settings should be possible.
I'd need to talk to James about how the multiple frame buffer stuff works though, and how adding a 3rd display could be represented through the various APIs.

jriwanek, 2019-01-18:
Thanks muchly for the replies and information, I understand completely that it'd be low priority as my remaining requested use case is very much niche due to it being only of use to us Compute Module folk, but I'll gladly give whatever help I can if y'all start prodding around with it.

jriwanek, 2019-01-18:
Oh, forgot to add: It's not touch on both DSI screens I'm concerned about, just being able to use two screens, touch on either or both would be nice, but is entirely optional.