raspberrypi/firmware #1095: [Question] Other Feature support

jriwanek, 2019-01-18:
1) Is it possible to add support/is it supported already to boot off the second SD card on a normal pi or Compute Module?
2) Any chance of adding codec key support for DTS/DDP as it appears to be mentioned in the firmware?
3) On the NEC CM3, Regions 44 and 45 appear to be used for Codec License keys, which is which, and would it be possible to have a config.txt option to burn (valid) license keys into the OTP for other models?
4) Any of the high rolling CM customers requested NAND and/or SPI boot support in the firmware yet?

JamesH65, 2019-01-23:
No answers, because I don't know the answers to any of the questions.

jriwanek, 2019-01-23:
Okies, thanks!

jriwanek, 2019-01-23:
@pelwell As you're about, anything you can say on the above? (Thanks in advance even if it's the same as James' response - I do appreciate all the time y'all put into answering my rather obviously tedious questions)

pelwell, 2019-01-23:
1. A quick question to my neighbour suggests that, although it is possible to make a device boot from GPIOs 227-27, this is an alternative (not an addition) controlled by an OTP bit, making the change irreversible.
2. No idea. @popcornmix ?
3. Valid licence keys are, once entered into config.txt, automatically burnt into OTP.
4. I don't know, and if I did I wouldn't tell you. ;-p

jriwanek, 2019-01-23:
1) Do you know the bit number/can you tell me?
3) That must have changed since I last looked, I'll give it a go later today
4) That's fine, can't blame a girl for asking!

5) specifically on 17-25, am I right that's related to program_emmc_boot=<blah>
6) I know from prior discussion y'all can't tell me a lot of the OTP stuff, but is it possible to tell me if I've gotten anything *wrong* on: <https://github.com/jriwanek/RaspberryPi/wiki/OTP>
Even if it's just "you misnamed block blah" or "you got that bit wrong" even if you can't tell me what the right answer is?

jriwanek, 2019-01-23:
Did literally just before you responded
1, 3, 4, 5, 6 in order.

pelwell, 2019-01-23:
1. Bit 22 of row 17 (and 18 for redundancy), but
    a) I'm not responsible if you brick your Pi messing with the OTP, whether the information is correct or incorrect, and
    b) I doesn't help you unless you have nefarious means to write to the OTP.
5. No.
6. 22 is not programmed by `program_gpio_bootmode=` - that's just affects 19&20.

pelwell, 2019-01-23:
No more questions - you are delaying the implementation of your feature request, and more important things.

jriwanek, 2019-01-23:
Thanks again for your time!

closed 2019-11-05 by JamesH65