raspberrypi/firmware #1096: RPI3B+ USB boot failure, Drive not detected by bootcode.bin

Ve2mrx, 2019-01-20:
Hi!

I'm trying to boot my rPi 3B+ from a USB-attached SSD in an external case.
Drive: Kingston ssdNow V200 128GB SSD
Case: Kingston-provided SNA-DC/U fw:1.08

After reading a few other posts about USB boot issues, I tried a few ways to get it to work:

1. Try to boot without an SD card: FAIL, Never comes up.

2. Try to boot with only bootcode.bin and "timeout" file on the SD card: FAIL, see attached file boot-failed-bootcode_bin-only.log

3. Try to boot with full Boot files on SD card: PASS, see attached file boot-success-allfiles.log 

Boot files where up to date as of this post (sudo rpi-update, and copied `/boot` to `/boot-sd`). I tried with nothing else than the drive connected to the Pi's USB ports, no change.

Related link: [Raspberry Pi boot modes](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/)

**What I would like:** To boot with minimal files on the SD card, and the rest on the SSD. 
Right now, ALL of `/boot` is mirrored on the SD card mounted as `/boot-sd` and it's working. I can live with that, but if it's supposed to work, I'm ready to help get it to work. 

Thanks!
Martin

```
ve2mrx@rPi:~ $ uname -a
Linux rPi 4.14.93-v7+ #1191 SMP Wed Jan 16 11:53:33 GMT 2019 armv7l GNU/Linux

ve2mrx@rPi:~ $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0 119.2G  0 disk
|-sda1        8:1    0  43.9M  0 part /boot
`-sda2        8:2    0  29.7G  0 part /
mmcblk0     179:0    0 970.5M  0 disk
`-mmcblk0p1 179:1    0 970.4M  0 part /boot-sd

ve2mrx@rPi:~ $ lsusb
Bus 001 Device 005: ID 11b0:6298 ATECH FLASH TECHNOLOGY
Bus 001 Device 004: ID 0424:7800 Standard Microsystems Corp.
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
[boot-failed-bootcode_bin-only.log](https://github.com/raspberrypi/firmware/files/2776264/boot-failed-bootcode_bin-only.log)
[boot-success-allfiles.log](https://github.com/raspberrypi/firmware/files/2776265/boot-success-allfiles.log)

lwpz, 2019-02-28:
@Ve2mrx Have you managed tried to put the /boot and img on the USB drive itself?

Ve2mrx, 2019-03-01:
I did not succeed. I just left /boot on a 1GB  SD and map it as such in /etc/fstab. 

My first reason was to prevent wear on the SD, and since the root is on the SSD, I am satisfied. As long as the SD card is mapped to /boot in fstab, everything works as designed!

Martin

servusDei2018, 2019-07-23:
Hmmm, I'm very interested in this discussion. @Ve2mrx , **may I please have instructions to do what you did?**

Like yourself, I have a SSD attached to my RPi 3b+. **I was able to get it to boot directly from the SSD (without any SD card)**, but last Friday (July 19), after I `sudo apt-get update` and `sudo apt-get upgrade`, my Pi refused to boot from the SSD :(

When copy a backup of `/boot` to the SSD, it boots, but there's lots of errors resulting from the old kernel without its modules -- `[FAILURE] Failed to start Load Kernel Modules`.

**I would appreciate any help to get my Pi booting either wholly or partially from my SSD again.**

Ve2mrx, 2019-07-23:
@servusDei2018 You need to modify your /etc/fstab to point /boot to the SD card, while keeping the root on the SSD. I also linked 

/etc/fstab:
```
proc            /proc           proc    defaults          0       0

# Boot-SD - Boot files on SD card, kept on SD for better compatibility
#PARTUUID=c68bbc05-01  /boot           vfat    defaults          0       2
**/dev/mmcblk0p1         /boot**          vfat    defaults          0       2

# Boot-SSD - Manual copy of /boot, residual of cloning my SD to the SSD - Unused by the Pi
#PARTUUID=a62ba183-01  /boot           vfat    defaults          0       2
/dev/sda1              /boot-ssd      vfat    defaults          0       2

# Root partition - Set by Unique ID, contains all other Linux files
**PARTUUID=_a62ba183-02_  /**               ext4    defaults,noatime,discard  0       1
#/dev/sda2              /               ext4    defaults,noatime,discard  0       1

# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
```
**Of course, the PARTUUID for the root partition on your SSD will be different, modify accordingly. You may want to have some other Linux machine nearby as a mistake in that file will prevent your Pi from booting.**

In order not to dilute this bug report and annoy the DEVs, anything NOT related to the issue mentioned in the issue title should taken elsewhere like a forum or PMs.

Thanks,
Martin