raspberrypi/firmware #1097: Latest Revision breaks Netboot

Flole998, 2019-01-20:
Updating to latest master breaks Netboot (it stops after requesting fixup.dat from TFTP). With 9baae7655f01ac37ba3b28c0053e9c6d0085dcfb it seems to work.

@ghollingworth Do you still have your test setup from the other issue? Could you run a quick test and see if you are experiencing the same issue?

Flole998, 2019-01-20:
9baae7655f01ac37ba3b28c0053e9c6d0085dcfb works perfectly fine, this is what I reverted to and it just booted and rebooted, everytime I update it stops working.
For reference, this is the working bootcode:
```
#sha1sum bootcode.bin
c43d24c2dbed7cad569e82395bfc29461fd3e2db  bootcode.bin
```

popcornmix, 2019-02-04:
bootcode.bin hasn't changed since that update.
Latest bootcode.bin has the same sh1sum.

Flole998, 2019-02-04:
Maybe it's the fixup or a different file that's causing the issues? I'll try later again with the latest to see if I'm still having the same issue

closed 2022-01-10 by Flole998