raspberrypi/firmware #1098: Overlay for SSD1306 framebuffer for OLED display.

mincepi, 2019-01-21:
I've successfully used the new SSD1307fb framebuffer driver with inexpensive SSD1306 controlled 128x32 and 128x64 OLED displays. They're available for a couple of dollars US from the usual Chinese suppliers, or from Adafruit as parts 3527 and 326.

Here's the overlay source:

// Overlay for SSD1306 128x64 and 128x32 OLED displays
/dts-v1/;
/plugin/;

/ {
    compatible = "brcm,bcm2718";

    fragment@0 {
	target = <&i2c1>;
	__overlay__ {
	    status = "okay";

	    #address-cells = <1>;
	    #size-cells = <0>;

	    ssd1306: oled@3c{
		compatible = "solomon,ssd1306fb-i2c";
		reg = <0x3c>;
		solomon,width = <128>;
		solomon,height = <64>;
		solomon,page-offset = <0>;
	    };
	};
    };

    __overrides__ {
	address = <&ssd1306>,"reg:0";
	width = <&ssd1306>,"solomon,width:0";
	height = <&ssd1306>,"solomon,height:0";
	offset = <&ssd1306>,"solomon,page-offset:0";
	normal = <&ssd1306>,"solomon,segment-no-remap?";
	sequential = <&ssd1306>,"solomon,com-seq?";
	remapped = <&ssd1306>,"solomon,com-lrremap?";
	inverted = <&ssd1306>,"solomon,com-invdir?";
    };

};

Also, here's a fragment that could be included in the overlay README:

Name:   ssd1306
Info:   Overlay for activation of SSD1306 over I2C OLED display framebuffer.
Load:   dtoverlay=ssd1306,<param>=<val>
Params: address                 Location in display memory of first character.
                                (default=0)
        width                   Width of display. (default=128)
        height                  Height of display. (default=64)
        offset                  virtual channel a. (default=0)
        normal                  Has no effect on displays tested. (default=not set)
        sequential              Set this if every other scan line is missing.
                                (default=not set)
        remapped                Set this if display is garbled. (default=not set)
        inverted                Set this if display is inverted and mirrored.
                                (default=not set)

        Examples:
        Typical usage for 128x64 display: dtoverlay=ssd1306,inverted

        Typical usage for 128x32 display: dtoverlay=ssd1306,inverted,sequential

        i2c_baudrate=400000 will speed up the display.

        i2c_baudrate=1000000 seems to work even though it's not officially supported
        by the hardware, and is faster still.

        For more information refer to the device datasheet at:
        https://cdn-shop.adafruit.com/datasheets/SSD1306.pdf

Feel free to include them in Raspbian if you wish: that would free me from maintaining my own copies!

I have a project page here if you want more info: sites.google.com/site/mincepi/pi2oled

Email me at mincepi -at- gmail.com with questions. If I need to do this request via a different method please let me know.

pelwell, 2019-01-21:
A Pull Request is the usual way to submit an overlay - include an entry in the Makefile (in alphabetical order, please), and a "Signed-off-by:" line - but I have enough here to work with if you'd rather not.

pelwell, 2019-01-21:
P.S. This all looks good - just watch the 80 character line length limit in the README (and don't forget the two blank lines between entries).

mincepi, 2019-01-21:
I'm avoiding learning pulls - for now at least. If you can work with what I
submitted I would appreciate it!

pelwell, 2019-01-21:
Can I add a Signed-off-by: line for you?

pelwell, 2019-01-21:
Done - https://github.com/raspberrypi/linux/commit/0126956700c2c8a2ff5ed8441e07046d5b7c508a

mincepi, 2019-01-21:
> Done - raspberrypi/linux@0126956

Are the .dtbo files automatically generated, or should I submit one?

pelwell, 2019-01-21:
The line added to the Makefile takes care of the compilation.

mincepi, 2019-01-21:
Ah, I see that now. Thanks for all your help!

ypopovych, 2024-01-27:
Hello @mincepi,

Seems as solomon driver has `reset-gpios` option, which allows to setup GPIO pin for screen reset.

Could you, please, add this option too, so it can be configured from config.txt?