raspberrypi/firmware #1100: RPI 3B+ state of /sys/class/net/eth0/carrier stuck at 1

brubbel, 2019-01-23:
TLDR; The eth0 carrier state flag gets stuck at '1' if /sys/class/net/eth0/carrier is polled within <1 second after unplugging the network cable.

**Hardware** (3 devices were tested, on 2 different networks):
1. RPI3B+ (rev. a020d3), new raspbian (kernel 4.14.79-v7+).
2. Raspberry pi official power supply (5.1V, 2.5A)
3. peripherals: ethernet cable, usb keyboard (dell), hdmi screen (dell).

**Setup**
1. OS is written to SD-card, ssh is not enabled.
2. ethernet cable is plugged in **_before_** boot.
3. local login.
4. cat /sys/class/net/eth0/carrier (**do enter yet!**)
5. unplug ethernet cable
6. enter within 1 second after unplug.

**Result**
The /sys/class/net/eth0/carrier flag gets stuck at '1'.

**Expected result**
The /sys/class/net/eth0/carrier flag becomes '0'

**Situation where this issue is relevant**
A script may be checking the carrier flag to failover to WiFi, for example every 10 seconds. However, every once in a while a race condition can occur where this bug is triggered on disconnecting the cable.

brubbel, 2019-01-23:
@JamesH65 Very possibly related:
https://github.com/raspberrypi/linux/issues/2617#issuecomment-405625812

XECDesign, 2019-01-28:
This is also affecting dhcpcd, which fails to switch over to wifi if the cable is unplugged.

In my case, I can reproduce this by booting with the ethernet cable plugged in and running `ip monitor link dev eth0`. Then unplugging or plugging the cable back in shows no activity.

If I boot with the cable unplugged, it works as expected.

The issue is present in 4.14 and 4.19. 

I tried a patch that adds `tasklet_schedule(&dev->bh);` back into `lan78xx_open`, but that made no difference.

brubbel, 2019-01-28:
One remarkable thing is that ETHTOOL_GLINK (SIOCETHTOOL ioctl) reports the correct link status, so it must be that the carrier status flag is not extracted from what the hardware reports in real time.

e.g. `ethtool eth0`

JamesH65, 2019-01-29:
Looking at https://github.com/raspberrypi/linux/blob/rpi-4.14.y/drivers/net/usb/lan78xx.c#L1179, this is the section of code that handles the link being dropped, I wonder if there needs to be some sort of flushing of the various SKB's etc that might be in flight. For example, there was a recent fix to the end of this function https://github.com/raspberrypi/linux/blob/rpi-4.14.y/drivers/net/usb/lan78xx.c#L1237 that makes sure requried processing is done when the link comes up. Complete guess though.

What i don;t understand is why the polling seems to kill it, that's just a read as far as I cna see, so would not expect that to cause this. Unless there's a mutex issue somewhere perhaps?

JamesH65, 2019-01-29:
I don't seem to be able to replicate this on 4.14.90 using any of the mechanisms above. Don't think there is anything unusual in my kernel setup, non-tainted, so a pure 4.14.90.

brubbel, 2019-01-29:
Bug is still there. Just tested with 4.14.90-v7+ (upgraded from 4.14.79-v7+).
Link speed is 1000Mb/s when connected, if this may be relevant.

When cable is unplugged:

`Linux raspberrypi 4.14.90-v7+ #1183 SMP Fri Dec 21 14:03:50 GMT 2018 armv7l GNU/Linux
`

```
Settings for eth0:
        Supported ports: [ TP MII ]
        Supported link modes:   10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Advertised link modes:  10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Full 
        Advertised pause frame use: Symmetric
        Advertised auto-negotiation: Yes
        Link partner advertised link modes:  10baseT/Half 10baseT/Full 
                                             100baseT/Half 100baseT/Full 
        Link partner advertised pause frame use: Symmetric Receive-only
        Link partner advertised auto-negotiation: Yes
        Speed: 100Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 1
        Transceiver: internal
        Auto-negotiation: on
        Supports Wake-on: pumbag
        Wake-on: g
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: no
```

```
pi@raspberrypi:~ $ cat /sys/class/net/eth0/carrier
1
```

brubbel, 2019-01-29:
Strange. Since the rpi-update (and downgrade to original kernel), carrier is always stuck at 1, unless I boot without cable. Will check a second RPI in a moment.

brubbel, 2019-01-29:
Second RPI (rpi-update to 4.14.94-v7+), cable is unplugged.

`cat /sys/class/net/eth0/carrier`
switches between
`cat: /sys/class/net/eth0/carrier: Invalid argument`
and
`1`

brubbel, 2019-01-29:
More info (back to fresh SD card). I have the impression that thing became worse after the rpi-update.

Boot with cable: carrier stuck (only unbind/bind the driver solves it)
Boot without cable: works! (carrier switches nicely as it should do)

pi@raspberrypi:~ $ cat /etc/rpi-issue 
Raspberry Pi reference 2018-11-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7e0c786c641ba15990b5662f092c106beed40c9f, stage2
pi@raspberrypi:~ $ vcgencmd version
Nov  4 2018 16:31:07 
Copyright (c) 2012 Broadcom
version ed5baf9520a3c4ca82ba38594b898f0c0446da66 (clean) (release)
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

XECDesign, 2019-01-29:
I doubt it makes a difference, but as a data point - I have wifi connected at the same time. I could also give you my sd card tomorrow if it's not reproducible on your setup.

brubbel, 2019-01-29:
Just tested: the **desktop** version (2018-11-13-raspbian-stretch.img) **does not show the problem**.
The lite version (2018-11-13-raspbian-stretch-lite.img) does.

I have the impression that 'something' is happening too fast after boot in the lite version, which can explain why an unbind/bind cycle solves the issue.
Could it be that the LAN7515 chip is not (completely) ready yet before the kernel initiates communication?

XECDesign, 2019-01-29:
I was using the desktop version.

brubbel, 2019-01-29:
I suppose it isn't possible to recompile/insmod the driver without recompiling the kernel?

XECDesign, 2019-01-29:
It is, but it's a bit of a hassle. Install raspberrypi-kernel headers and compile the module like it's out of tree. https://www.kernel.org/doc/Documentation/kbuild/modules.txt

brubbel, 2019-03-18:
Happens also on 3B+, "Raspbian Stretch desktop 2018-11-13", 4.14.98-v7+.
Network does not (always) switch to WiFi when cable is unplugged. Manual `ifconfig eth0 down` solves this.
I didn't look further into the matter yet though.

maxnet, 2019-04-11:
I have a theory that this is caused by that:
1) the phy code tells the NIC to generate link status change notification "interrupts" right away on boot.  lan78xx_probe() -> lan78xx_phy_init() -> phy_connect_direct() -> phy_start_interrupts() -> lan88xx_phy_config_intr()
2) however it only opens USB communication to listen for and handle such "interrupts" much later when the network interface is brought up (by dhcpcd) and lan78xx_open() gets called

By the time it gets to 2 the phy has already detected carrier, and it missed the interrupt about it.
Since the interrupt was not acknowledged, it does not send any new ones.

Seems to work better with the following hack:

```
diff -ur linux-rpi-4.19.y.orig/drivers/net/usb/lan78xx.c linux-rpi-4.19.y/drivers/net/usb/lan78xx.c
--- linux-rpi-4.19.y.orig/drivers/net/usb/lan78xx.c	2019-04-10 14:31:00.000000000 +0200
+++ linux-rpi-4.19.y/drivers/net/usb/lan78xx.c	2019-04-11 17:19:04.360246346 +0200
@@ -2692,6 +2692,9 @@
 				  "intr submit %d\n", ret);
 			goto done;
 		}
+		/* ack any previous interrupts we may have missed. */
+		if (net->phydev->drv->ack_interrupt)
+			net->phydev->drv->ack_interrupt(net->phydev);
 	}
 
 	lan78xx_init_stats(dev);
```

But will leave a proper fix to someone else.
Not sure what the CLEAN way to handle this is.

MichaelM223, 2019-07-03:
Is this being worked on at the moment?
I'm having problems with the link going down not being detected and hence the Pi (3B+) won't failover to a 3G dongle. (default route for eth0 does not change).

Built a kernel with the 'hack' mentioned above for the lan78xx module but that does not change anything either unfortunatly.

Rubs1er, 2020-08-12:
has anyone found a solution?
/sys/class/net/eth0/carrier stuck in 1 if boot with cable
also I notice that
after down and up eth0 
/sys/class/net/eth0/carrier stuck in 0

5.4.51-v7+

pelwell, 2020-12-16:
A possible fix for this is in the current rpi-update kernel, which has just switched to rpi-5.10.y.

closed 2020-12-19 by brubbel

ruimo, 2021-04-23:
Hi, I found exactly the same issue still existing and rpi-update does not work. Any suggestions?

Procedure:
1) Burn SD card using rpi-imager, selected other => Raspberry Pi OS Lite (32-bit).
2) Turn on Raspberry Pi with burned SD.
3) ping raspberrypi.local works correct.
4) Unplug the ethernet cable.
5) Wait a minute.
6) Plug the ethernet cable back.
7) ping raspberrypi.local will not work any more. Ping IP where the IP shown in 3) also does not work.

Environment:
Raspberry Pi 3
Raspberry Pi OS Lite (32bit)

rpi-update shows the following version:
 *** depmod 5.10.31-v7+                
 *** depmod 5.10.31-v7l+                              
 *** depmod 5.10.31-v8+                                    
 *** depmod 5.10.31+