raspberrypi/firmware #1101: bootcode.bin doesn't implement pxe-service correctly

alkisg, 2019-01-24:
Hi, this line is needed in dnsmasq.conf for raspberry pi clients to boot:
> pxe-service=X86PC,"Raspberry Pi Boot"

A line like this would be better:
> pxe-service=ARM32_EFI,"Raspberry Pi Boot",raspberrypi/config.txt

X86PC means "a x86 computer with BIOS". ARM32_EFI is a new option (http://manpages.ubuntu.com/manpages/cosmic/en/man8/dnsmasq.8.html), but a literal "10" can be used in older dnsmasq versions.

The second change, "raspberrypi/config.txt" is more important. It allows the server to define a path, a subdirectory, where the firmware lives. This helps in organization, when the TFTP dir is used for other things than netbooting raspberries, but more importantly, it allows to specify "this raspberry lab will boot with raspbian/config.txt, and the other lab will boot with ubuntu-mate/config.txt", i.e. multiple images for different clients.

popcornmix, 2019-02-01:
ping @ghollingworth

ghollingworth, 2019-02-01:
This is the wrong place for this.  It is not an issue with the firmware.  Maybe you want the documentation changed?  In which case provide a pull request there.

ghollingworth, 2019-02-01:
Ah I see, it can be changed but will only effective if you're using an SD card with bootcode.bin on it

alkisg, 2019-02-01:
That will be awesome, thanks! :)

alkisg, 2020-01-12:
Any progress on this? I'm in the process of adding support for Raspberries in the ltsp.org netbooting software, and it's a mess to have to copy all files to the TFTP root directory instead of subdirectories...

ghollingworth, 2020-01-12:
We can't fix this in the bootrom (since it's ROM not firmware), so can only be changed if booted with an SD card.  But this would then mean dnsmasq.conf would need to be different for the two different modes to boot correctly.

If you want to boot two different Pis with different images you can do that using the serial number sub-directory instead.  you can then provide symbolic links in your tftp directories pointing to the right build files the same way as we do it with PiServer

closed 2020-01-12 by ghollingworth

alkisg, 2020-01-12:
Hi @ghollingworth,

In a x86 Linux installation, please run `ls /boot/grub`. Would you like to have all these files in the root / directory? It would be a mess... That's why bootloaders use subdirectories like /boot/grub or /EFI, especially in shared resources like a root file system or a root TFTP directory.

Currently in my system I have 15 subdirectories in TFTP, one for each different architecture or distribution. Only Raspberry is broken and needs 15 files in the TFTP root dir; all the rest architectures play nice.

Regarding the migration period, it can be like all migration periods. If a boot filename is defined in pxe-service, then the firmware can use it, and fall back to / if it's not there:

> pxe-service=ARM32_EFI,"Raspberry Pi Boot",raspberrypi/config.txt

If someone has clients with both the old and the new firmware, he can keep using the root TFTP dir. If one has only SD cards or if 5 years later has has only new rpis with the new firmware, then he can finally have an organized TFTP root. Typical migration periods, nothing special.

In addition, by allowing a subdirectory there, a sysadmin can then easily split the clients from the DHCP config, without having to create a mess of symlinks based on client MAC/IP addresses. I.e. it's a lot easier to offer a TFTP/raspbian and a TFTP/ubuntu-mate, without having to use symlinks for each one of e.g. 100 clients.

Is the problem the proposal itself or the related work? Is that part of the code open? Do you accept patches?

Thanks,
Alkis

lurch, 2020-01-17:
> Is that part of the code open?

AFAIK the net-booting code is all inside bootcode.bin / the ROM on the BCM2837 SoCs / the (beta) EEPROM image on the Pi4 - all of which are closed-source.

skinkie, 2022-02-01:
@ghollingworth I do hope this issue (PXE advertising itself as X86) is reported upstream.

For dnsmasq I have solved the issue in this way:
```
dhcp-match=set:x86PC,option:client-arch,0 # Legacy BIOS
dhcp-match=set:BC_EFI,option:client-arch,7 # BC EFI
dhcp-match=set:BC_EFI,option:client-arch,9 # BC EFI
dhcp-match=set:ARM,option:client-arch,10 # BC EFI
dhcp-match=set:ARM64,option:client-arch,11 # BC EFI

dhcp-mac=set:PI4,dc:a6:32:*:*:*

pxe-service=tag:PI4, 0, "Raspberry Pi Boot"
dhcp-boot=tag:BC_EFI,syslinux.efi
dhcp-boot=tag:x86PC,lpxelinux.0
```

lurch, 2022-02-02:
@skinkie If you're exclusively using the Pi4, there's a bunch of PXE-related issues in https://github.com/raspberrypi/rpi-eeprom/search?q=PXE&type=issues
You may find that your issue has already been discussed, but if not I guess you could open a new one?

skinkie, 2022-02-02:
> @skinkie If you're exclusively using the Pi4, there's a bunch of PXE-related issues in https://github.com/raspberrypi/rpi-eeprom/search?q=PXE&type=issues You may find that your issue has already been discussed, but if not I guess you could open a new one?

Did not know that there was another repo for that :)

timg236, 2022-02-02:
The issue has already been raised there. There's no plan to change this because none of the other options match either.

A better method would be to use DHCP option 97 because this identifies the board-type as well
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#DHCP_OPTION97https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#DHCP_OPTION97

skinkie, 2022-02-02:
> There's no plan to change this because none of the other options match either.

What about fixing all things that are obviously _wrong_.

timg236, 2022-02-02:
> > There's no plan to change this because none of the other options match either.
> 
> What about fixing all things that are obviously _wrong_.

Changing the default behaviour for a large number of installed computers would be a bad and unnecessary change. 
This setting could be overridden by the EEPROM config allowing Pi4 to behave differently to existing Raspberry Pis. That would be a pretty low priority change though since you'd need to update the bootloader on existing boards.

skinkie, 2022-02-02:
> Changing the default behaviour for a large number of installed computers would be a bad and unnecessary change. This setting could be overridden by the EEPROM config allowing Pi4 to behave [...]

Lets say that the values now cannot be used at all. So that's why I think it is valuable.