raspberrypi/firmware #1103: SMBus Block Read

AntonioSanta, 2019-01-29:
Hello
I have seen the issue 
https://github.com/raspberrypi/firmware/issues/828 
but anyway I still can find that recent kernel doesn't provide support for SMBus read block.
See attachement for output from I2cdetect -F 1.
In the overlay folder I can see there is I2c1-bcm2708.dtbo . 
am I missing something I should do to enable this SMBus read block ?
thank you
[image]

pelwell, 2019-01-29:
Please add `dtoverlay=i2c1-bcm2708,combine=off` and retest.

AntonioSanta, 2019-01-29:
by adding this line

_dtoverlay=i2c1-bcm2708,combine=off_

as last line on the config.txt in the Boot folder I get the same result from i2cdetect -F 1.
_SMBus Block Read            no_

I am using this system : 

[image]

AntonioSanta, 2019-01-29:
thank you for the link, that's interesting, which file is that define in ?
so previous  https://github.com/raspberrypi/firmware/issues/828 is not possible to solve as well, I thought from those messages that it was possible in some way.
do you know if it is because of an hardware limitation of the I2c controller compatibility with SMBus block read or if it is just because it was chosen to not implement it ?
thanks

dognotdog, 2019-10-03:
I just wanted to add that `read_i2c_block_data()` might work in lieu of `read_block_data()` despite the SMBus block read not being supported if you know the number of bytes to expect, and don't have to rely on the controller detecting it. The basic structure of the read looks to be the same on the i2c bus, except that the `count` field is ignored by the hardware/driver and passed on to the application. (caveat: I've only tested this with an MCU slave that hasn't been tested against a proper SMBus controller)