raspberrypi/firmware #1104: Connect mysql server remotely by php api

ram94638, 2019-01-31:


pelwell, 2019-01-31:
Your issue:
1. sounds like a question, without being a question,
2. is nothing to do with the firmware, and
3. would be better posed as a question in the [Raspberry Pi Forums](https://www.raspberrypi.org/forums/index.php).

closed 2019-01-31 by pelwell

ram94638, 2019-01-31:
I have hosted website on Ubuntu LAMP. Now I want to manage that My wrbsite should I am able to access mysql server by ssh, mysqlWorkbench and other accesable tool. But not through php code.

> Your issue:
> 
> 1. sounds like a question, without being a question,
> 2. is nothing to do with the firmware, and
> 3. would be better posed as a question in the [Raspberry Pi Forums](https://www.raspberrypi.org/forums/index.php).

Yup,

ram94638, 2019-01-31:
> Your issue:
> 
> 1. sounds like a question, without being a question,
> 2. is nothing to do with the firmware, and
> 3. would be better posed as a question in the [Raspberry Pi Forums](https://www.raspberrypi.org/forums/index.php).

yup