raspberrypi/firmware #1105: tvservice -n does not detect display name but dumps EDID correctly

marcus905, 2019-02-01:
Hello,

I am working with a Raspberry Pi 3B+ and a 1024x600 touchscreen interfaced through HDMI and USB through a TI TFP401 decoder that needs a special config.txt line for it to work properly, as the TFP401 doesn't support scaling. I wanted to add an `[EDID=...]` filter in config.txt but to no avail as `tvservice -n` returns

`[E] No device present`

all the while EDID is read and parsed correctly through `tvservice -d` then decoded through `edid-decode` to validate correctness.

The display is powered by the Raspberry Pi itself through USB and this apparently is the same issue as reported in #401 but with the workaround of the `bootcode_delay=10` entry in config.txt is not working as the USB hub is not powered (enough?) in bootcode apparently and it is therefore not reading the EDID.

Powering the display separately through a dedicated PSU makes everything work ok on the EDID side but since the display has a touchscreen and it is powered via the same USB cable as the touchscreen transmits data on, powering the display separately is not a very desirable option as it would entail not using the touchscreen at all.

Is there any chance that there could be a config.txt entry to early-power the USB in bootcode too, or at least power the USB before the delay is taken into consideration?
Alternately, would it be possible to move the EDID detect phase later in the bootcode, after the USB powering event?
Do you have any other suggestion to help with this?

Thank you!

pelwell, 2019-02-01:
A 3B+ requires a 2.5A supply to function well on its own. Expecting it to power an external display sounds optimistic. I think you should power the display directly.

pelwell, 2019-02-01:
I don't think we'll be redesigning the startup sequence for your application - getting the filters to work as expected is already tricky - but @popcornmix may be feeling more charitable.

marcus905, 2019-02-01:
Thank you, I'd really appreciate that a lot!

I think that the code that manages this is not open source, else I can try helping you out if you'd like that.

popcornmix, 2019-02-01:
> Powering the display separately through a dedicated PSU makes everything work ok on the EDID side but since the display has a touchscreen and it is powered via the same USB cable as the touchscreen [...]

You may have to show a picture, but normally if you want to provide separate power from data for a USB device you use a Y cable. e.g. https://www.amazon.co.uk/StarTech-feet-Cable-External-Drive-Black/dp/B003HHK576

marcus905, 2019-02-01:
I will try with a Y cable as a stopgap measure and report back.