raspberrypi/firmware #1106: Legacy (non-GL desktop driver) causing incorrect tensorflow.js tf.sqrt() calculation result.

bertlea, 2019-02-05:
Tensorflow.js guys told me a bug I raised to them should be a bug in the firmware (of Raspberry Pi).  Basically, I found that when I use tensorflow.js on Chromium in "webgl" mode, its tf.sqrt() returns a wrong value (very close to zero).  That issue only exist if I use the default "Legacy, non-GL driver", it will not have issue if I configure the Pi use experimental GL driver).

More details on how to reproduce the issue and some screenshots can be found in the link:

[https://github.com/tensorflow/tfjs/issues/1073](url)

popcornmix, 2019-02-05:
There is no webgl support when using the legacy gl driver.
You need to use arm side driver for that.

bertlea, 2019-02-06:
@popcornmix Do you mean it is the Tensorflow.js mistakenly identify system can do webgl?  The Tensorflow.js should detect if browser supports webgl or not, and if not, it shall fallback to use CPU.  In this case, the Tensorflow mistakenly using webgl.... but other functions works fine, just the square-root tr.sqrt() not working.    Thus, I still think it is the issue of the driver that seems doing software emulation of GPU for webgl, but got a bug on sqrt() function.

popcornmix, 2019-02-06:
All I can say if the browser will not be using the firmware gl driver for webgl.
There is no plumbing that can make that possible.

As such it's not a firmware issue.