raspberrypi/firmware #1107: Mailbox interface get/set voltages issue

tester962, 2019-02-06:
I'm trying to get/set the voltage using the Mailbox interface, but that seems to be broken in the firmware.
The wiki (https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface) says that the values are "offset from 1.2V in units of 0.025V", but I don't think that's true anymore. 
The returned values seem to be the voltage in microvolts.

But these voltages don't match the values obtained using the "vcgencmd" tool. 
For "over_voltage = 0" the voltage should be 1.20V, but I get 1.312500V from the mailbox. For "over_voltage = 2" the voltage should be 1.25V, but I get 1.356V.

SDRAM voltages seem to be reported correctly (1.20V).

Also, how am I supposed to set the voltages dynamically? No matter what voltage values I try, the mailbox returns value 200000000.

JamesH65, 2019-02-06:
I'm partway through moving the mailbox information from the wiki to actually documentation. Been updating stuff as I go, but sure if that section is any different though

https://github.com/raspberrypi/documentation/blob/JamesH65-mailbox_docs/configuration/mailboxes/propertiesARM-VC.md#voltage

tester962, 2019-02-06:
The voltage is measured in microvolts, not millivolts.

The mailbox interface seems to return wrong voltage values. 
Also, what kind of value does the "Set voltage" property expect?

popcornmix, 2019-02-07:
> For "over_voltage = 0" the voltage should be 1.20V, but I get 1.312500V from the mailbox. For "over_voltage = 2" the voltage should be 1.25V, but I get 1.356V.

This isn't true. We run AVS on bootup that measures a ring oscillator to determine speed of the silicon and that determines the default voltage. ``over_voltage`` values are then added to this default.

On Pi0/Pi1 the default voltage is always 1.2V, but on Pi2/Pi3 it varies per board (between 1.2V and 1.4V).

So I think software is correct but documentation may not match.

tester962, 2019-02-07:
Yes, the documentation seems to be the issue there.

How can I set the voltages using the mailbox interface? I've tried requesting a voltage (in microvolts), but I keep getting value "200000000".

popcornmix, 2019-02-07:
In general the voltage is constantly adjusted by the firmware (and when switching turbo mode) so this isn't a supported option.
Can you say why you want to do this?

tester962, 2019-02-07:
I'm changing the ARM clock frequency dynamically (including some minor overclock) and increasing the voltage will make the system more stable. While running at lower frequencies (like 600MHz) I'd like to decrease the voltage.

popcornmix, 2019-02-07:
This isn't something that is supported by current firmware.

tester962, 2019-02-07:
OK, thank you for the clarification.

closed 2019-02-07 by tester962