raspberrypi/firmware #1111: help with omx video_render only showing first frame when using OMX_DISPLAY_ROT90

zoff99, 2019-02-18:
here is the testcase we use:

https://gist.github.com/zoff99/ff0dad829b912f12308db2bbf3d8a5d1

if we change OMX_DISPLAY_ROT90 to OMX_DISPLAY_ROT0
we get a nice animation of frames. otherwise only the first frame is displayed.

what mistake did we make? any help welcome

zoff99, 2019-02-19:
we updated the testcase.
as soon as the rotation is activated the animation/video stops, after disabling the rotation is moves again.

please help

6by9, 2019-02-19:
Your test case is doing invalid buffer handling as you are always sending st->buffer[0] into OMX_EmptyThisBuffer.

Once you pass a buffer off to OMX_EmptyThisBuffer (or OMX_FillThisBuffer) you MUST wait for the OMX_EmptyThisBufferDone or OMX_FillThisBufferDone callback before touching that buffer again, and particularly before resubmitting that same buffer. Buffers are stored in linked lists, so being presented with the same buffer again easily results in a loop in the list -> GPU stuck in a loop. (There is actually some handling to stop this from actually occurring).

- When not transposing, the buffer presented to video_render is the actual memory block that is rendered. If you update the buffer again then those differences are reflected immediately on the display.
- When transposing the transpose has to be done first to another buffer which is then displayed. Because you're passing the same buffer multiple times, the system isn't seeing any updates, and therefore won't retrigger the transpose into the live buffer.

6by9, 2019-02-19:
It looks like you'd started to implement the correct handling, and then didn't.
Replace the end of omx.c with 
```
int omx_display_input_buffer(struct omx_state *st,
                             void **pbuf, uint32_t *plen)
{
    int buf_idx;

    if (!st->buffers)
    {
        return EINVAL;
    }
    buf_idx = st->current_buffer;

    *pbuf = st->buffers[buf_idx]->pBuffer;
    *plen = st->buffers[buf_idx]->nAllocLen;
    st->buffers[buf_idx]->nFilledLen = *plen;
    st->buffers[buf_idx]->nOffset = 0;

    return 0;
}

int omx_display_flush_buffer(struct omx_state *st, uint32_t data_len)
{
    int buf_idx = st->current_buffer;
    st->current_buffer = (st->current_buffer+1) % st->num_buffers;

    st->buffers[buf_idx]->nFlags = OMX_BUFFERFLAG_STARTTIME;
    st->buffers[buf_idx]->nOffset = 0;
    st->buffers[buf_idx]->nFilledLen = data_len;
    st->buffers[buf_idx]->nTimeStamp = omx_ticks_from_s64 (0);

    if (OMX_EmptyThisBuffer(st->video_render, st->buffers[buf_idx])
            != OMX_ErrorNone)
    {
        dbg(9, "OMX_EmptyThisBuffer error\n");
    }

    return 0;
}
```
And it'll update correctly.
This still isn't totally correct as you are making the assumption that the buffer will have been returned, not actually checking that it has.

You're also doing further things wrong as:
- if you hit ctrl-c then it takes a long time to quit. The GPU is logging an error on natural program exit.
- Each run is leaking GPU memory (```sudo vcdbg reloc``` shows "RIL buffer" allocations from each run). I see no calls to ```OMX_FreeBuffer```, nor ```OMX_FreeHandle```, so this is not unexpected.

strfry, 2019-02-19:
Thanks @6by9  for promptly fixing my broken code. We owe you a few beers sometime. :)

Apparently i never bothered to implement the multi-buffer handling, because the buffers never queued up without the rotation.
Just one more thing, how can i access the GPU log buffer the see the mentioned error?

6by9, 2019-02-19:
The buffers didn't queue up as part of the implementation of IL on the Pi is that EmptyThis Buffer will memcpy the buffer you provide into a mirror buffer within the GPU's memory, and that happens before the check of whether the buffer has been doubly submitted. So the GPU buffer got updated regardless, and then it would have returned the buffer once to ensure counts tallied.
If there is any processing required on the buffer by the component then double submission will have issues.

```vcgencmd set_logging level=0x100c0```, run your app, and then ```sudo vcdbg log msg``` gives one level of logging. (The 0xc0 is to enable the IL level logging).
The logging messages are all from the firmware side so use it as guidance only as the source code is closed source.

The other one is to add ```start_debug=1``` to /boot/config.txt, and then ```sudo vcdbg log assert``` will report asserts that fire. Again you won't be able to refer back to the source as to the exact cause of the assert, but generally the assert condition will give you a fairly good clue.
In this case I'm seeing
```
805384.017: assert( comp->port[i].nHostBuffers == 0 && comp->port[i].vc_buffer.pool == ILCS_NUM_BUFFER_POOLS && comp->port[i].host_buffer.pool == ILCS_NUM_BUFFER_POOLS ) failed; ../../../../../applications/vmcs/ilcs/ilcs_in.c::ilcs_component_deinit line 247 rev 4e7f7d7
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
805384.162: assert( seen == total ) failed; ../../../../../applications/vmcs/ilcs/ilcs_in.c::ilcs_component_deinit line 272 rev 4e7f7d7
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
```
I haven't verified with the debugger, but the basic summary is that there are host buffers outstanding at the point that the component is being destroyed.

zoff99, 2019-02-19:
@6by9 if you have a BTC address we can send you some beer money

6by9, 2019-02-19:
> if you have a BTC address we can send you some beer money

No need. I'm employed by RPT therefore this is part of my job.

zoff99, 2019-02-19:
@6by9 alright, mate. you helped us on 3 issues every time in less than 1 day.
keep up the good work.

6by9, 2019-02-20:
If you're happy that this issue is resolved, then please close it.

zoff99, 2019-02-20:
we will run some tests hopefully today. and then close it.

closed 2019-02-20 by zoff99