raspberrypi/firmware #1113: Limit NAL size/camera crash

Thalhammer, 2019-02-21:
I'm trying to limit the size of NAL packets generated by the h264 encoder.
I'm using OpenMAX (via https://github.com/gagle/raspberrypi-omxcam).
I already tried 3 different options with varying success:
* OMX_IndexConfigVideoNalSize
  The "offical" way by Khronos
* OMX_IndexConfigBrcmVideoEncodedSliceSize
  The equivalent of Broadcom
* OMX_IndexConfigBrcmVideoEncoderMBRowsPerSlice
  Another option that should control slice sizes

None of those work.
I added my code here:
https://github.com/gagle/raspberrypi-omxcam/blob/master/src/h264.c#L200
similar to the way all other options are set.

**OMX_IndexConfigVideoNalSize**
Seems like this is not supported by the firmware (OpenMAX returns a not supported error).

**OMX_IndexConfigBrcmVideoEncodedSliceSize**
Tried various values ranging from 1k upto 100k but either they crash the camera or they don't affect the output (i.e. the nals are the same size even if the set value is lower).

**OMX_IndexConfigBrcmVideoEncoderMBRowsPerSlice**
Tried 2 and 4 both of which crash the camera.

When the camera crashes I get a single buffer of 29 bytes, after which it just hangs and if I kill the program the camera can not be opened a second time (OMX_ErrorInsufficientResources / mmal_vc_component_enable: failed to enable component: ENOSPC) until I reboot my Pi.
Which makes trying different values quite a challenge.

Is this a firmware bug or am I missing something ?

6by9, 2019-02-21:
```OMX_IndexConfigBrcmVideoEncoderMBRowsPerSlice``` is the same as MMAL_PARAMETER_MB_ROWS_PER_SLICE. Support for that was added to raspicam with https://github.com/raspberrypi/userland/pull/486/files. Whilst integrating that we discovered that something limits encoding to 1280 wide if using more than one slice per frame. Sorry, this isn't a priority for us to investigate, so it is what it is.
Note this is the number of rows of macroblocks per NAL slice, therefore a value of 2 would be 320 slices per 720p frame which is silly.
Whilst this may reduce the size of the NALs it will negatively effect the compression efficiency and latency as each slice has to be compressed independently.
Note there is also ```OMX_IndexParamBrcmNALSSeparate``` to force each NAL to come out as an independent IL buffer.

I can't recall what ```OMX_IndexConfigBrcmVideoEncodedSliceSize``` actually does, and the docs don't help much. I'll check when I get to the office.

Fundamentally the video_encode process is a single pass process. The compression parameters are all set before the frame is started based on the encoded sizes produced by previous frames and target bitrate. Should you get a scene change or other big difference in the source stream at an inconvenient moment then a P-frame will end up being very large, with subsequent frames being small or even dropped to compensate.
IIRC In some modes there is a path where it will just drop the frame should it exceed a buffer size threshold (I can't recall how to configure this). The issue with this is that the encode is pipelined so that CABAC is independent of motion pred etc, therefore the encode of next frame has started using the current frame as a reference before the final size of the current frame has been determined.

You say that 
> I'm trying to limit the size of NAL packets generated by the h264 encoder.

What are you actual requirements and why? Below a specific size, or just generally reduce them? For what purpose?

Thalhammer, 2019-02-21:
> Note this is the number of rows of macroblocks per NAL slice, therefore a value of 2 would be 320 slices per 720p frame which is silly.

Ah so I interpreted it wrong, though it would be rows per slice (i.e. 2 = 310 cols).

> I can't recall what `OMX_IndexConfigBrcmVideoEncodedSliceSize` actually does.

It limits the size of SLICEs to a certain number of bytes ( and since IDR_SLICE and SLICE are the biggest in a stream, it effectivly limits the nal size to this value + overhead). This value is exactly what I need and as far as I can tell it works as long as the resolution is 1280x720 or lower, which is consistent with what you said about `OMX_IndexConfigBrcmVideoEncoderMBRowsPerSlice`.

> Note there is also `OMX_IndexParamBrcmNALSSeparate` to force each NAL to come out as an independent IL buffer.

Yeah i figured this out already, thought it was the default, as in all my tests with 1080p I got exactly 1 buffer per NAL. It works fine, I was just to stupid to set it.

> What are you actual requirements and why? Below a specific size, or just generally reduce them? For what purpose?

I have a size restriction of about 2500 byte per nal. It's the MTU of my data link and keeping the nal size below it allows me to fit 1 nal per packet and thus remove a lot of complexity from both sending and receiving side, as well as reduce latency (No packet reassembly, no FEC calculation). And as the packet size is quite low it also limits the distortion in case a packet is lost.

6by9, 2019-02-21:
>>    Note this is the number of rows of macroblocks per NAL slice, therefore a value of 2 would be 320 slices per 720p frame which is silly.

> Ah so I interpreted it wrong, though it would be rows per slice (i.e. 2 = 310 cols).

Sorry, typo by me. Yes, 2 would be 310 slices, each 2 macroblocks high (32 pixels as macroblocks are 16x16) .

I think your interpretation of ```OMX_IndexConfigBrcmVideoEncodedSliceSize``` in that it splits the frame into slices at when it hits that size. There's a comment in the code
```
   //! Set a hard limit on the slice (/video packet/network packet) size.  Default of 0 implies no limit
   //! Note that for H.264, this switches on auto_slice_mode.
```

The issue is within the firmware. I hadn't previously logged an issue on the slicing width limit at 1280 beyond the comments on https://github.com/raspberrypi/userland/pull/519 and https://github.com/raspberrypi/userland/pull/486. I have now done so with #1114.