raspberrypi/firmware #1114: video_encode slice mode limited at 1280 pixels wide

6by9, 2019-02-21:
(Note for myself as much as anything)

As noted when https://github.com/raspberrypi/userland/pull/486 was being merged, using H264 sliced mode encoding the codec stalls should the width be > 1280 pixels.
Cause unknown at present, but it's buried down within venc.

Thalhammer, 2019-02-21:
When trying to use 1080p + nalsize limit the following appears in the gpu log (vcdbg log msg);
```
872846.111: VCE taking >1s to run. Status=0x11100a0 PCs=(0x24ac,0x24b0,0x24a0,0x249c) bad_addr=0x0
872846.204: Last VCE launch code: 0x1ec27778 for 13296 data: 0x1ec2ab68 for 7704
```

Maybe it helps.

6by9, 2019-02-21:
The VCE is one of the hardware blocks, and has timed out. Why is the bigger question.

Sorry, this really isn't one that those outside of RPT can assist with, and it isn't a priority as slice mode is a very niche area.

daleiliu, 2020-08-09:
From my understanding, this feature is critical to h.264 rtp packets.
As defined in [RFC 6184: RTP Payload Format for H.264 Video](https://tools.ietf.org/html/rfc6184), 

> 6.2.  Single NAL Unit Mode
>
>   This mode is in use when the value of the OPTIONAL packetization-mode
>   media type parameter is equal to 0 or the packetization-mode is not
>   present.  All receivers MUST support this mode.

If the encoder does not support OMX_IndexConfigBrcmVideoEncodedSliceSize in FHD mode, the application will not be able to generate Single NAL Unit packets and may cause interoperability issues with some RTP receiver, like SIP endpoints and video conference applications.

6by9, 2020-08-09:
> From my understanding, this feature is critical to h.264 rtp packets.

As per the part you quote, it is critical for the **receiver** (ie decoder) to be able to support Single NAL unit mode.
There is no obligation on the sender (encoder) to use it as packetization-mode is listed as optional. The other packetisation modes are listed as SHOULD and MAY be supported by the receiver respectively, therefore we're in the one mode that is mandated to work with the receiver.

I'm not clear whether there is any issue anyway. The whole frame is encoded as a single NAL unit. Section 6 is all dealing with packetisation rules (ie an RTP process, not an encoder). Even if the frame was encoded in multiple NALs, then the rules still apply to each of those NAL units.

daleiliu, 2020-08-11:
Thanks for your reply.  Yes I agree it's critical to the receiver.  My point is, if the RPi-based sender can send the packets without specifying "packetization-mode=1" in SDP, it will have a bigger chance to talk to the other end because the RFC requires the support.  From my observation, some hardware SIP endpoints do not support "packetization-mode=1" explicitly in SDP offers and may cause SDP negotiation issues.

Another minor issue is, in case there is some packet loss, one missing RTP packet may cause the whole frame to be lost if it's encoded as one single NAL unit, while in the other way, it will only show some bad blocks.

daleiliu, 2020-08-11:
For example, the SDP office sent by an application, with the current limitation, has to specify two offers:
1. h264 profile, packetization-mode=0, 1280x720
2. h264 profile, packetization-mode=1, 1920x1080 (or even higher resolution)

If the receiver supports packetization-mode=0, 1920x1080, at the end the video conversation can only use 1280x720.