raspberrypi/firmware #1115: graphics_get_display_size always return -1

FranciscoMoya, 2019-02-23:
I made a tiny test to invoke graphics_get_display_size from bcm_host. It returns -1 for any display.

Tested on Raspberry Pi 3B, fully updated raspbian stretch, updated firmware, hdmi to vga converter. 

Kivy egl backend depends on this.

Am I the only one to experience this?  Is there any further info I can provide to track this down?

Regards,

pelwell, 2019-02-23:
Are you calling bcm_host_init first to establish the connection to the firmware?

The quickest route to a solution is to provide the source of the simplest test that demonstrates the problem.

mutantbob, 2020-05-23:
```import ctypes

bcm = ctypes.CDLL('libbcm_host.so')

eglint = ctypes.c_int

b = bcm.bcm_host_init()
assert b==0
width=eglint()
height = eglint()

s = bcm.graphics_get_display_size(0,ctypes.byref(width),ctypes.byref(height))
print(s)
```
gives me -1 as well on my pi