raspberrypi/firmware #1116: program_gpio_bootmode fails to boot with pull-ups applied

johntomko, 2019-02-27:
Hello,

I'm using a Pi3 A+  with latest firmware. I've followed documentation from https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/bootflow.md and successfully disabled boot from all sources (no pull-ups applied).

I now have a pull-up applied to GPIO 22 according to documentation (3.3V, 4.7K resistor. Tested with several different resistor values as well to no avail), which I expect to enable boot from primary SD, but with the SD card inserted the Pi does not boot. It is powered (red PWR LED on) but I see no boot activity on the ACT LED. I've also pulled down all other relevant GPIOs

I haven't tried enabling UART debugging yet as per https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/ so that is a possible next step I suppose. 

This has me really stumped, any help would be appreciated.

pelwell, 2019-03-01:
It seems the pin assignments changed with 2837B0. Before then, the roles were:

    22 -> BRCM SD
    23 -> Arasan SD
    24 -> NAND
    25 -> SPI EEPROM
    26 -> USB device

With 2837B0 the roles are:

    20 -> BRCM SD
    21 -> Arasan SD
    22 -> NAND
    23 -> SPI EEPROM
    24 -> USB device
    25 -> USB host MSD
    26 -> USB host Ethernet

johntomko, 2019-03-01:
SD boot successful! USB boot didn't work right away, but the drive could need re-imaged. I'll test that next and report back ASAP. In the meantime, thanks for the help!

johntomko, 2019-03-02:
Update: SD boot was successful on pin 20, but USB boot was unsuccessful on pins 24, 25, and 26. I was able to boot via USB using program_usb_boot_mode prior to enabling GPIO boot. 

I noticed in the comments at https://www.raspberrypi.org/blog/raspberry-pi-3-model-bplus-sale-now-35/ there was reference to some bugfixes between A0 and B0 revisions, could this have something to do with that?

johntomko, 2019-03-04:
Edited the issue. After some more research and an opt_dump I found what was happening with USB, and both were able to successfully boot independently. Pin 20 with SD, pin 25 with a USB host MSD.

However, with both inserted, both pin 20 and 25 boot form the SD (different file structure on each device, the SD card's files are present after boot). I have to remove the SD entirely for the USB device to boot, even with it selected via GPIO.

pelwell, 2019-03-04:
Is this with a pull-up on GPIO 20? If so, that's what I would expect, since SD takes precedence over USB.

johntomko, 2019-03-04:
That's with a pull-up on 25, 20 is grounded. It almost looks like the USB device starts the boot process (with its bootcode.bin, etc.) but the file system for the SD is still loaded. That seems to me like it's selecting boot devices properly, but it still uses the default boot hierarchy for loading files.

pelwell, 2019-03-04:
Ah - I think you're using a different definition of "boots from" than I was expecting. For clarity, please state where the following are loaded from in your scenario:
* bootcode.bin
* start.elf
* kernel*.bin, etc.

johntomko, 2019-03-04:
Those are being loaded from the USB host MSD, selected by pin 25. It appears that is done correctly.

Using SSH, I connect to the Pi and the file structure at the root directory is that of the SD card, not the USB device.

pelwell, 2019-03-04:
The kernel has an independent view of where the rootfs is, as controlled by cmdline.txt. You can specify it either by UUID (e.g. `root=PARTUUID=abcd1234-02`) or by device (e.g. `root=/dev/sda1`).

johntomko, 2019-03-04:
I see, the USB device is configured to look for the SD card's rootfs. Is it worth it to make a new USB image from scratch, or is there some UUID I can find to manually change that setting? I imaged the USB device with a custom SD image, I'm guessing a fresh Raspbian install would resolve the issue.

pelwell, 2019-03-04:
There's no need for a new image - just edit the cmdline.txt's `root=` to refer to the chosen device or partition. You can retrieve the 8-digit volumeid with `sudo blkid /dev/sda0` (or however the USB driver appears), then set `root=PARTUUID=<volumeid>-<partition-index>`, where `<partition-index>` is the number after the `p`, e.g. `/dev/sda0p2` has an index of 2.

johntomko, 2019-03-04:
Success! I'm closing this issue, that did indeed work Both boot modes are working properly, thanks for the help!

closed 2019-03-04 by johntomko