raspberrypi/firmware #1117: Wifi AP broken with driver nl80211 between 4.14.90 (maybe after) and 4.19.25

ccleaud, 2019-02-28:
Hi all,

Here's a bug report for you.
I hope this helps you to fix.
Thanks for you work.

Regards,
C.CLEAUD

-----

## Problem description

Yesterday (2019-02-27) I updated my `Raspberry Pi 3 B Rev 1.2` firmware, using `rpi-update`.
This gets the most recent commit from <https://github.com/Hexxeh/rpi-firmware/commits/master>, which is a mirror to your repo <https://github.com/raspberrypi/firmware/commits/master>.
This installed the firmware as in commit f84820ed7fcfd0f0d364e3d47cdbc83215596b2c (from your repo).

You need to known my Pi was configured to act as a Wifi Access Point.
After restarting, I noticed Wifi Access Point didn't work anymore.
I tried to changes Wifi-AP settings but the service `hostapd` always failed to start.
When manually trying to start `hostapd`, I got an error (see later).

So, after Googling, I decided to downgrade firmware to the last release of yeaf 2018.
Running command `rpi-update da5948d8ef8354557732d9c8f5ad5e7e24374a69` installed the firmware as in commit e1bd9b0b8cda901ee9b23cbb8b3334cde71320a7 (from your repo).
After reboot, the Wifi AP got back. No setting file was changed.

My Wifi settings use `nl80211` driver. So, something is getting broken with this since 2018-12-21.

For security reasons, I've obfuscated my device name and wifi settings.

## Symptoms

```text
# service hostapd status
hostapd.service - LSB: Advanced IEEE 802.11 management daemon
   Loaded: loaded (/etc/init.d/hostapd; generated; vendor preset: enabled)
   Active: active (exited) since Wed 2019-02-27 21:05:29 UTC; 3s ago
     Docs: man:systemd-sysv-generator(8)
  Process: 2177 ExecStop=/etc/init.d/hostapd stop (code=exited, status=0/SUCCESS)
  Process: 2212 ExecStart=/etc/init.d/hostapd start (code=exited, status=0/SUCCESS)

Feb 27 21:05:29 xxxxxxxx systemd[1]: Starting LSB: Advanced IEEE 802.11 management daemon...
Feb 27 21:05:29 xxxxxxxx hostapd[2212]: Starting advanced IEEE 802.11 management: hostapd failed!
Feb 27 21:05:29 xxxxxxxx systemd[1]: Started LSB: Advanced IEEE 802.11 management daemon.
```

## Manual start attempt

Please note the `nl80211: Register frame command failed (type=208): ret=-22 (Invalid argument)`

```text
# hostapd -d /etc/hostapd/hostapd.conf
random: Trying to read entropy from /dev/random
Configuration file: /etc/hostapd/hostapd.conf
rfkill: initial event: idx=0 type=1 op=0 soft=0 hard=0
rfkill: initial event: idx=1 type=2 op=0 soft=0 hard=0
nl80211: Using driver-based roaming
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Using driver-based off-channel TX
nl80211: Supported vendor command: vendor_id=0x1018 subcmd=1
nl80211: Use separate P2P group interface (driver advertised support)
nl80211: Enable multi-channel concurrent (driver advertised support)
nl80211: use P2P_DEVICE support
nl80211: Disable use_monitor with device_ap_sme since no monitor mode support detected
nl80211: interface wlan0 in phy phy0
nl80211: Set mode ifindex 3 iftype 3 (AP)
nl80211: Setup AP(wlan0) - device_ap_sme=1 use_monitor=0
nl80211: Subscribe to mgmt frames with AP handle 0x134a038 (device SME)
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x134a038 match=

nl80211: Register frame command failed (type=208): ret=-22 (Invalid argument)

nl80211: Register frame match - hexdump(len=0): [NULL]

nl80211: Could not configure driver mode

nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
nl80211: Remove beacon (ifindex=3)
netlink: Operstate: ifindex=3 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211: Set mode ifindex 3 iftype 2 (STATION)
nl80211: Teardown AP(wlan0) - device_ap_sme=1 use_monitor=0

nl80211 driver initialization failed.

hostapd_interface_deinit_free(0x1348cd8)
hostapd_interface_deinit_free: num_bss=1 conf->num_bss=1
hostapd_interface_deinit(0x1348cd8)
wlan0: interface state UNINITIALIZED->DISABLED
hostapd_bss_deinit: deinit bss wlan0
wlan0: AP-DISABLED 
hostapd_cleanup(hapd=0x13499a0 (wlan0))
hostapd_free_hapd_data: Interface wlan0 wasn't started
hostapd_interface_deinit_free: driver=(nil) drv_priv=(nil) -> hapd_deinit
hostapd_interface_free(0x1348cd8)
hostapd_interface_free: free hapd 0x13499a0
hostapd_cleanup_iface(0x1348cd8)
hostapd_cleanup_iface_partial(0x1348cd8)
hostapd_cleanup_iface: free iface=0x1348cd8
```

## Context of /etc/hostapd/hostapd.conf

Here is `vi /etc/hostapd/hostapd.conf` :

```text
      1 # --- Hardware settings ---
      2 # --------
      3 interface=wlan0
      4 
      5 ssid=XXXXXXXX
      6 
      7 # Recommended channels 802.11b: 1, 6, 11, 14
      8 # Recommended channels 802.11g/n: 1, 6, 11
      9 channel=1
     10 
     11 # Beacon interval in kus (1.024 ms)
     12 beacon_int=512
     13 
     14 
     15 
     16 # --- WPA and WPA2 configuration ---
     17 # --------
     18 # Station MAC address -based authentication (driver=hostap or driver=nl80211)
     19 # 0 = accept unless in deny list, one MAC per line in the form 99:99:99:99:99:99
     20 # 1 = deny unless in accept list, one MAC per line in the form 99:99:99:99:99:99
     21 # 2 = use external RADIUS server (accept/deny lists are searched first)
     22 macaddr_acl=0
     23 
     24 # Accept/deny lists are read from separate files
     25 #accept_mac_file=/etc/hostapd.accept
     26 #deny_mac_file=/etc/hostapd.deny
     27 
     28 # Indicates algorithm specified by IEEE 802.11
     29 auth_algs=1
     30 
     31 # AP will broadcast SSID
     32 ignore_broadcast_ssid=0
     33 
     34 # --- WPA settings ---
     35 # WPA algorithm used (WPA2 in this case)
     36 wpa=2
     37 wpa_passphrase=XXXXXXXXXX
     38 wpa_key_mgmt=WPA-PSK
     39 wpa_pairwise=TKIP
     40 rsn_pairwise=CCMP
     41 
     42 
     43 
     44 # --- Hardware configuration ---
     45 # --------
     46 # Type of driver to be used (in may be different depending on your WiFi dongle chipset)
     47 # In majority of cases it will be 'driver=nl80211'
     48 driver=nl80211
     49 
     50 # Whether IEEE 802.11n (HT) is enabled
     51 # Note: You will also need to enable WMM for full HT functionality.
     52 # Note: hw_mode=g (2.4 GHz) and hw_mode=a (5 GHz) is used to specify the band.
     53 ieee80211n=1
     54 
     55 # User-friendly description of device (optional)
     56 device_name=RTL8192CU
     57 
     58 # Manufacturer name (optional)
     59 manufacturer=Realtek
     60 
     61 # WPS RF Bands (a = 5G, b = 2.4G, g = 2.4G, ag = dual band)
     62 hw_mode=g
     63 
     64 # Enable WMM
     65 wmm_enabled=1
     66 
     67 # Enable 40MHz channels with 20ns guard interval
     68 ht_capab=[HT40][SHORT-GI-20][DSSS_CCK-40]
     69 
```

6by9, 2019-02-28:
(This is not a firmware issue and should really be reported in https://github.com/raspberrypi/linux/issues)

Slight difference as it's referencing the onboard wi-fi of the 3B+, but
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=224931&start=100#p1434854
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=224931&start=25#p1396409
is similar.
Can you try updating hostapd? You'll have to build it from source.

What exact make and model is your wifi dongle? It's very hard to tell what chipset is inside a dongle without having one, so if we're to try and reproduce then we need exactly the same module.

popcornmix, 2019-02-28:
I belive buster hostapd is new enough. It should be possible to install this through [debian backports](https://backports.debian.org/).

popcornmix, 2019-02-28:
Note: we are planning to update hostapd in raspbian soon, but confirmation that it fixes your problem would be helpful.

XECDesign, 2019-02-28:
hostapd is not in stretch-backports, but I've build the packages here: 
https://www.dropbox.com/s/qdy7jridugqkx7y/wpa_stretch.tar.xz?dl=1

ccleaud, 2019-03-01:
Hi again,

Sorry for the delay. Also sorry for the test: it doesn't work.
I still have to give you the exact make and model of my wifi dongle.

# Step 0 - initial (and working) state

My R-Pi firmware is at commit e1bd9b0b8cda901ee9b23cbb8b3334cde71320a7 and hostapd is working.

# Step 1 - downgrade firmware, keep hostapd

```sh
rpi-update 82f49d8dfd9c000a203fb1deae64a3eb87abf3c1
reboot
```

Test hostapd version:

```sh
hostapd -v
```

```text
hostapd v2.4
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2015, Jouni Malinen <j@w1.fi> and contributors
```

Start hostapd:

```sh
hostapd -dd /etc/hostapd/hostapd.conf
```

```text
random: Trying to read entropy from /dev/random
Configuration file: /etc/hostapd/hostapd.conf
rfkill: initial event: idx=0 type=1 op=0 soft=0 hard=0
rfkill: initial event: idx=1 type=2 op=0 soft=0 hard=0
nl80211: Using driver-based roaming
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Using driver-based off-channel TX
nl80211: Supported vendor command: vendor_id=0x1018 subcmd=1
nl80211: Use separate P2P group interface (driver advertised support)
nl80211: Enable multi-channel concurrent (driver advertised support)
nl80211: use P2P_DEVICE support
nl80211: Disable use_monitor with device_ap_sme since no monitor mode support detected
nl80211: interface wlan0 in phy phy0
nl80211: Set mode ifindex 3 iftype 3 (AP)
nl80211: Setup AP(wlan0) - device_ap_sme=1 use_monitor=0
nl80211: Subscribe to mgmt frames with AP handle 0x1cf5038 (device SME)
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x1cf5038 match=
nl80211: Register frame command failed (type=208): ret=-22 (Invalid argument)
nl80211: Register frame match - hexdump(len=0): [NULL]
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
nl80211: Remove beacon (ifindex=3)
netlink: Operstate: ifindex=3 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211: Set mode ifindex 3 iftype 2 (STATION)
nl80211: Teardown AP(wlan0) - device_ap_sme=1 use_monitor=0
nl80211 driver initialization failed.
hostapd_interface_deinit_free(0x1cf3cd8)
hostapd_interface_deinit_free: num_bss=1 conf->num_bss=1
hostapd_interface_deinit(0x1cf3cd8)
wlan0: interface state UNINITIALIZED->DISABLED
[... 151 lines cut ...]
wlan0: AP-DISABLED
wlan0: Unable to setup interface.
hostapd_interface_deinit_free(0x201cf8)
hostapd_interface_deinit_free: num_bss=1 conf->num_bss=1
hostapd_interface_deinit(0x201cf8)
wlan0: interface state DISABLED->DISABLED
hostapd_bss_deinit: deinit bss wlan0
wlan0: Deauthenticate all stations
nl80211: sta_remove -> DEL_STATION wlan0 ff:ff:ff:ff:ff:ff --> 0 (Success)
wlan0: AP-DISABLED
hostapd_cleanup(hapd=0x202a48 (wlan0))
hostapd_free_hapd_data: Interface wlan0 wasn't started
hostapd_interface_deinit_free: driver=0xe7448 drv_priv=0x2035b0 -> hapd_deinit
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
nl80211: Remove beacon (ifindex=3)
netlink: Operstate: ifindex=3 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
hostapd_interface_free(0x201cf8)
hostapd_interface_free: free hapd 0x202a48
hostapd_cleanup_iface(0x201cf8)
hostapd_cleanup_iface_partial(0x201cf8)
hostapd_cleanup_iface: free iface=0x201cf8
```

-> It still fails with many lines `nl80211: Register frame command failed (type=208): ret=-22 (Invalid argument)`

ccleaud, 2019-03-01:
Some more details about my Pi, even if the problem is solved using a newer version of hostapd:

```sh
lshw
```

```text
XXXXXXXX
    description: ARMv7 Processor rev 4 (v7l)
    product: Raspberry Pi 3 Model B Rev 1.2
    width: 32 bits
    capabilities: smp
[...]
  *-network:0
       description: Wireless interface
       physical id: 2
       logical name: wlan0
       serial: XX:XX:XX:XX:XX:XX
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=brcmfmac driverversion=7.45.98.38 firmware=01-e58d219f ip=172.16.0.1 multicast=yes wireless=IEEE 802.11
```

XECDesign, 2019-03-05:
The packages have been uploaded to the stretch repo.

locinus, 2019-10-11:
Hi. Snoopy noob question here (not familiar with the apt-get world):
in @ccleaud solution, I block on:

`dpkg -i hostapd_2.6-21~bpo9~rpt1_armhf.deb`

=> `cannot access archive: No such file or directory` (or `Unable to locate package` if I try apt-get install).
I suppose it's a repository that I haven't set on my Raspberry? (though I see it in /etc/apt/sources.list.d/raspi.list)

Could you tell in details how to actually install that very package?

p1g30n, 2019-12-05:
@locinus You can download the file from here https://archive.raspberrypi.org/debian/pool/main/w/wpa/

AntonElsesser, 2021-08-09:
Is there any update to this topic, because I have the same problem with the latest image and updates.
Also a downgrade not helped for me.

WIFI Adapter

Bus 001 Device 004: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter

hostapd -v
hostapd v2.8-devel
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors

root@raspberrypi:~# hostapd -dd /etc/hostapd/hostapd.conf
random: getrandom() support available
Configuration file: /etc/hostapd/hostapd.conf
**nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0**
netlink: Operstate: ifindex=3 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211: Set mode ifindex 3 iftype 2 (STATION)
nl80211: Failed to set interface 3 to mode 2: -19 (No such device)
nl80211 driver initialization failed.
hostapd_interface_deinit_free(0x1e8f020)
hostapd_interface_deinit_free: num_bss=1 conf->num_bss=1
hostapd_interface_deinit(0x1e8f020)
wlan0: interface state UNINITIALIZED->DISABLED
hostapd_bss_deinit: deinit bss wlan0
wlan0: AP-DISABLED
hostapd_cleanup(hapd=0x1e8fda0 (wlan0))
wlan0: CTRL-EVENT-TERMINATING
hostapd_free_hapd_data: Interface wlan0 wasn't started
hostapd_interface_deinit_free: driver=(nil) drv_priv=(nil) -> hapd_deinit
hostapd_interface_free(0x1e8f020)
hostapd_interface_free: free hapd 0x1e8fda0
hostapd_cleanup_iface(0x1e8f020)
hostapd_cleanup_iface_partial(0x1e8f020)
hostapd_cleanup_iface: free iface=0x1e8f020