raspberrypi/firmware #1118: arm64 binaries

laroche, 2019-03-02:
Is it possible to provide binaries like vcgencmd and vcdbg
for aarch64?

Thanks a lot,

Florian La Roche

6by9, 2019-03-02:
We don't really support 64bit at present, although some work has been done for 64bit kernel support. 64bit userspace is still a way off.

vcgencmd is part of the userland repo - https://github.com/raspberrypi/userland/tree/master/host_applications/linux/apps/gencmd
You'll want https://github.com/raspberrypi/userland/pull/507 too if trying to make use the buildme script.

vcdbg isn't public, but it may be possible to make such a 64bit binary.

laroche, 2019-03-02:
Thanks for the pointer to vcgencmd.

I know 64bit is way off to be fully supported, but I'd be happy to see even unsupported
binaries to try them out.

I've just tried copying the 32bit libs/binaries, but seems this is not enough with plain
vanilla 64bit Debian from https://wiki.debian.org/RaspberryPi3:
# ./vcdbg log assert
debug_sym: OpenVideoCoreMemoryFileWithOffset: Unable to open '/dev/vc-mem': No such file or directory(2)
Unable to open videocore memory access: -2

So I'll anyway have to dig into a modified kernel first...

best regards,

Florian La Roche

6by9, 2019-03-02:
vc-mem is under the config BCM2708_VCMEM. I fixed the driver up for 64bit recently, and I thought I'd added it to bcmrpi3_defconfig but can't see it there now. Add it and that bit should work.
It's a pretty trivial driver that just allows key parameters to be passed from the firmware to vcdbg.

vcdbg will then try to use part of the bcm2708_fb driver to memcpy from GPU memory to ARM memory, but reverts to accessing via /dev/mem if that fails. Annoyingly it used to log a message every time it failed, so the useful stuff got partially hidden behind a load of annoying error messages. That should have been fixed recently.
It uses this bcm2708_fb quirk as there is 16MB of RAM that the GPU can access that the ARM can't. Should the relevant bit of information be in there then the ARM can't get to it.

satmandu, 2019-10-02:
Compiling the userland binaries works great except when one needs to run vcdbg.

Any chance of an arm64 binary of vcdbg being released so we don't have to juggle the libraries which are commong to vcdbg and other tools like vcgencmd?

satmandu, 2020-01-24:
For those who want to get vcdbg working on arm64, this should fix that, at least on arm64/ubuntu eoan and later:

(This assumes you have already compiled and installed the rest of the userland for arm64, and of course the binary would need a kernel compiled with  BCM2708_VCMEM.)
```
sudo apt install libgcc1-armhf-cross libstdc++6-armhf-cross libc6-armhf-cross patchelf -y
cd /usr/arm-linux-gnueabihf/lib
sudo curl -OL https://github.com/Hexxeh/rpi-firmware/raw/master/vc/hardfp/opt/vc/lib/libdebug_sym.so
sudo curl -OL https://github.com/Hexxeh/rpi-firmware/raw/master/vc/hardfp/opt/vc/lib/libelftoolchain.so
sudo curl -OL https://github.com/Hexxeh/rpi-firmware/raw/master/vc/hardfp/opt/vc/lib/libvcos.so
sudo mkdir -p /opt/vc/bin/
cd /opt/vc/bin/
sudo curl -OL https://github.com/Hexxeh/rpi-firmware/raw/master/vc/hardfp/opt/vc/bin/vcdbg
sudo chmod +x /opt/vc/bin/vcdbg
sudo cp /opt/vc/bin/vcdbg /opt/vc/bin/vcdbg.orig
sudo patchelf --force-rpath --set-rpath "/usr/arm-linux-gnueabihf/lib" /opt/vc/bin/vcdbg
sudo patchelf --set-interpreter /usr/arm-linux-gnueabihf/lib/ld-linux-armhf.so.3 /opt/vc/bin/vcdbg
```

satmandu, 2020-01-24:
@6by9 I have noticed that on pi4/arm64/ubuntu I'm able to get the vcdbg messages to show, but not on a 3b+. Both running (current) 5.3.0-1017-raspi2.

On Raspberry Pi 3 Model B Plus Rev 1.3:
```
cat /proc/version
Linux version 5.3.0-1017-raspi2 (buildd@bos02-arm64-063) (gcc version 9.2.1 20191008 (Ubuntu 9.2.1-9ubuntu2)) #19-Ubuntu SMP Thu Jan 16 18:25:50 UTC 2020
sudo vcdbg log msg 2>&1 >/dev/null | pastebinit
https://paste.ubuntu.com/p/nHVvxJNqc3/
```
On Raspberry Pi 4 Model B Rev 1.1:
```
cat /proc/version
Linux version 5.3.0-1017-raspi2 (buildd@bos02-arm64-063) (gcc version 9.2.1 20191008 (Ubuntu 9.2.1-9ubuntu2)) #19-Ubuntu SMP Thu Jan 16 18:25:50 UTC 2020
sudo vcdbg log msg 2>&1 >/dev/null | pastebinit
https://paste.ubuntu.com/p/rdG3gd2rDj/
```

Of note, the appropriate kernel driver appears to be compiled in:
```
grep -3 BCM2708_VCMEM /boot/config-5.3.0-1017-raspi2 
# Character devices
#
CONFIG_BRCM_CHAR_DRIVERS=y
CONFIG_BCM2708_VCMEM=y
CONFIG_BCM_VCIO=y
# CONFIG_BCM_VC_SM is not set
CONFIG_BCM2835_DEVGPIOMEM=y
```

6by9, 2020-01-24:
I did tweak libdebugsym.so to look at /proc/cmdline for the vc_mem parameters should /dev/vc-mem not exist, mainly as I wanted to be able to get logging out from a mainline kernel.
That appears not to have appeared in the userland repo yet though, but the version in this repo should include that. Rebuilding userland may have overwritten that though.

satmandu, 2020-01-24:
> I did tweak libdebugsym.so to look at /proc/cmdline for the vc_mem parameters should /dev/vc-mem not exist, mainly as I wanted to be able to get logging out from a mainline kernel.
> That appears not [...]

I'm pulling down vcdbg straight from the hexxeh repository, and using the same version on two different systems though. Are you saying that libdebugsym.so is the issue? Is there a known good version?

pelwell, 2020-01-25:
The version in the Hexxeh repo was updated 19 days ago: https://github.com/Hexxeh/rpi-firmware/blob/master/vc/hardfp/opt/vc/lib/libdebug_sym.so

vfazio, 2021-04-26:
Since the 64bit beta has been out for nearly a year, are there any updated timelines on providing a 64bit binary for vcdbg? patching the 32bit binary to work is a little hacky and while there is a 32bit static build for vcdbg here https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/67#issuecomment-653209729 it's not "maintained".

If there's a more appropriate issue tracking this, please let me know.

Sylensky, 2023-05-26:
I would like to bump this issue after 2 years where 64bit is much more common then before and wanted to know if this is possible by now?

popcornmix, 2023-05-26:
vclog is the replacement for vdbg and is available [here](https://github.com/raspberrypi/utils/tree/master/vclog).
64-bit vclog (and vcgencmd) are available from RPiOS 64-bit.

I think this issue can be closed.

laroche, 2023-05-26:
I agree. And if new issues should come up, their should go into new issue reports. I'll close this request.

best regards and thanks for your good/great engineering support,

Florian La Roche

closed 2023-05-26 by laroche