raspberrypi/firmware #1119: Debian spelling warnings

laroche, 2019-03-02:
Just looked more detailed into Debian binary packages and saw that the following
checks about spelling errors are part of the Debian raspi3-firmware package:

https://sources.debian.org/src/raspi3-firmware/1.20190215-1/debian/raspi3-firmware.lintian-overrides/

spelling-error-in-binary usr/lib/raspi3-firmware/start.elf reenable re-enable
spelling-error-in-binary usr/lib/raspi3-firmware/start.elf postion position
spelling-error-in-binary usr/lib/raspi3-firmware/start.elf intial initial
spelling-error-in-binary usr/lib/raspi3-firmware/start.elf Recyling Recycling
spelling-error-in-binary usr/lib/raspi3-firmware/start.elf rountine routine
spelling-error-in-binary usr/lib/raspi3-firmware/start.elf recieved received
spelling-error-in-binary usr/lib/raspi3-firmware/start.elf erorr error

spelling-error-in-binary usr/lib/raspi3-firmware/start_db.elf postion position
spelling-error-in-binary usr/lib/raspi3-firmware/start_db.elf reenable re-enable
spelling-error-in-binary usr/lib/raspi3-firmware/start_db.elf trucate truncate
spelling-error-in-binary usr/lib/raspi3-firmware/start_db.elf calulated calculated
spelling-error-in-binary usr/lib/raspi3-firmware/start_db.elf intial initial
spelling-error-in-binary usr/lib/raspi3-firmware/start_db.elf Recyling Recycling
spelling-error-in-binary usr/lib/raspi3-firmware/start_db.elf rountine routine
spelling-error-in-binary usr/lib/raspi3-firmware/start_db.elf recieved received
spelling-error-in-binary usr/lib/raspi3-firmware/start_db.elf erorr error
spelling-error-in-binary usr/lib/raspi3-firmware/start_db.elf preceeding preceding

spelling-error-in-binary usr/lib/raspi3-firmware/start_x.elf postion position
spelling-error-in-binary usr/lib/raspi3-firmware/start_x.elf reenable re-enable
spelling-error-in-binary usr/lib/raspi3-firmware/start_x.elf trucate truncate
spelling-error-in-binary usr/lib/raspi3-firmware/start_x.elf calulated calculated
spelling-error-in-binary usr/lib/raspi3-firmware/start_x.elf intial initial
spelling-error-in-binary usr/lib/raspi3-firmware/start_x.elf Recyling Recycling
spelling-error-in-binary usr/lib/raspi3-firmware/start_x.elf rountine routine
spelling-error-in-binary usr/lib/raspi3-firmware/start_x.elf recieved received
spelling-error-in-binary usr/lib/raspi3-firmware/start_x.elf erorr error

I haven't even checked if the problems are still present or if this is older cruft, but just wanted
to pass this along. Feel free to close this bug and then resolve this internally once you
have a lot of free time. ;-)

Let me add here to say a big thanks for the really good support for raspberry pi
from your side.

best regards,

Florian La Roche

pelwell, 2019-03-02:
Thanks for these. As you've acknowledged, fixing these typos is not going to be a high priority (and I'd push back on reenable which is acceptable), but we should get round to them eventually.

diederikdehaas, 2021-06-30:
Friendly ping?