raspberrypi/firmware #1121: Specifying hdmi_group=0 in config.txt

llks, 2019-03-14:
I have a display module with custom display timing (not the standard DMT or CEA) and i would like to use the "hdmi_group=0" in the "config.txt" to let the Raspberry Pi to do the "Auto-detect from EDID" which the EDID information is resided in the display module. However, it seems like the "hdmi_group=0" is not working this way. Has anybody successfully use "hdmi_group=0"("Auto-detect from EDID") in the "config.txt"? 

"hdmi_group=0" information of "Auto-detect from EDID" from: 
https://www.raspberrypi.org/documentation/configuration/config-txt/video.md

6by9, 2019-03-14:
Most people rely on the EDID reading, although it sounds like you;re relying on the detailed timings section.

What does the EDID actually report?
```tvservice -d edid.bin``` and post edid.bin

llks, 2019-03-14:
I am trying to set the display according to the display specification,
720x1280p hfp:80 hs:10 hbp:150 vfp:20 vs:10 vbp:20 pixel clock:76 MHz

I have read the EDID, please see it below, 

pi@raspberrypi:~ $ tvservice -d edid.bin
Written 128 bytes to edid.bin
pi@raspberrypi:~ $ edidparser edid.bin
Enabling fuzzy format match...
Parsing edid.bin...
HDMI:EDID version 1.3, 0 extensions, unknown aspect ratio
HDMI:EDID features - videodef 0x80 !standby !suspend !active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format is native; does not support GTF
HDMI:EDID found monitor name descriptor tag 0xfc
HDMI:EDID monitor name is T779-720x1280t?
HDMI:EDID found monitor range descriptor tag 0xfd
HDMI:EDID monitor range offsets: V min=0, V max=0, H min=0, H max=0
HDMI:EDID monitor range: vertical is 20-120 Hz, horizontal is 1-255 kHz, max pixel clock is 290 MHz
HDMI:EDID monitor range does not support GTF
HDMI:EDID failed to find a matching detail format for 720x1280p hfp:80 hs:10 hbp:150 vfp:20 vs:10 vbp:20 pixel clock:76 MHz
HDMI:EDID calculated refresh rate is 60 Hz
HDMI:EDID guessing the format to be 720x1280p @60 Hz
HDMI:EDID found unknown detail timing format: 720x1280p hfp:80 hs:10 hbp:150 vfp:20 vs:10 vbp:20 pixel clock:76 MHz
HDMI:EDID failed to find a matching detail format for 720x1280p hfp:80 hs:10 hbp:150 vfp:20 vs:10 vbp:20 pixel clock:76 MHz
HDMI:EDID calculated refresh rate is 60 Hz
HDMI:EDID guessing the format to be 720x1280p @60 Hz
HDMI:EDID found unknown detail timing format: 720x1280p hfp:80 hs:10 hbp:150 vfp:20 vs:10 vbp:20 pixel clock:76 MHz
HDMI:EDID established timing I/II bytes are 00 00 00
HDMI:EDID standard timings block x 8: 0x0101 0101 0101 0101 0101 0101 0101 0101
HDMI:EDID adding mandatory support for DMT (4) 640x480p @ 60Hz
HDMI:EDID filtering formats with pixel clock > 162 MHz or h. blanking > 1023
HDMI:EDID no known preferred format has been set
HDMI:EDID filtering preferred group has been changed from Invalid to DMT
HDMI:EDID best score mode initialised to DMT (4) 640x480p @ 60 Hz with pixel clock 25 MHz (score 0)
HDMI:EDID best score mode is now DMT (4) 640x480p @ 60 Hz with pixel clock 25 MHz (score 36864)
HDMI:EDID preferred mode is updated to DMT (4) 640x480p @ 60 Hz with pixel clock 25200000 Hz
HDMI:EDID has only DVI support and no audio support
edidparser exited with code 0
pi@raspberrypi:~ $

I see the system complain some "unknown detail timing format", is it we need other setting to use the "hdmi_group=0"?

6by9, 2019-03-14:
Please post the raw edid.bin file, not an analysis of it.

6by9, 2019-03-14:
Actually I think this is a duplicate of https://github.com/raspberrypi/firmware/issues/722
Specifying HDMI timings via EDID detailed timings is not supported.