raspberrypi/firmware #1123: vcgencmd measure_clock v3d spuriously returns 0

hhromic, 2019-03-15:
Hello, I noticed that the current firmware is returning spurious measurements of `0` for the `v3d` clock when the GPU is under stress. This doesn't seem to happen with other clocks. This is happening on RPI3B and RPI3B+.

For example, when running a GPU-intensive application such as RetroArch with an emulator and in another console measuring the `v3d` clock, you can observe this behaviour:
```
$ while true; do vcgencmd measure_clock v3d; sleep 0.1; done
frequency(43)=0
frequency(43)=0
frequency(43)=0
frequency(43)=0
frequency(43)=0
frequency(43)=250000000
frequency(43)=0
frequency(43)=0
frequency(43)=250000000
frequency(43)=250000000
frequency(43)=250000000
frequency(43)=250000000
frequency(43)=0
frequency(43)=250000000
frequency(43)=0
frequency(43)=250000000
frequency(43)=250000000
frequency(43)=300000000
frequency(43)=300000000
frequency(43)=250000000
frequency(43)=249999000
frequency(43)=250000000
frequency(43)=0
frequency(43)=0
frequency(43)=0
```

If you stop the GPU intensive task, the measurements are randomly left always at `0` or the actual clock, until you restart the GPU task and it goes oscillating again. I don't think this is a userland problem (the `vcgencmd` command itself) because other clocks are reported correctly and vcgencmd simply communicates with VC to obtain the clocks.

The packages on my system are all current:
```
ii  libraspberrypi-bin              1.20190215-1                      armhf        Miscellaneous Raspberry Pi utilities
ii  libraspberrypi-dev              1.20190215-1                      armhf        EGL/GLES/OpenVG/etc. libraries for the Raspberry Pi's VideoCore IV (headers)
ii  libraspberrypi0                 1.20190215-1                      armhf        EGL/GLES/OpenVG/etc. libraries for the Raspberry Pi's VideoCore IV
ii  raspberrypi-bootloader          1.20190215-1                      armhf        Raspberry Pi bootloader
ii  raspberrypi-kernel              1.20190215-1                      armhf        Raspberry Pi bootloader
```

ghollingworth, 2019-03-15:
The GPU does require the clock to be stopped between frames, so you're probably just catching it during   that time.

6by9, 2019-03-15:
Is this with the GL driver active? (dtoverlay=vc4-kms-v3d in /boot/config.txt)

The kernel driver does power the v3d block up and down as it is required to save power. The firmware driver leaves it on near constantly as we generally aren't that fussed over small power savings.

A quick test here using hello_fft with the firmware shows the v3d clock bumping to 300MHz when active, but otherwise 250MHz.
With vc4-kms-v3d, it drops back to 0, and turns up to 250 or 300MHz as I start glxgears, with the odd bump when idle.

hhromic, 2019-03-15:
@6by9 no I'm not using the GL driver, my config.txt file does not use the vc4-kms-v3d overlay.

I tried now with `hello_fft` and indeed it is not oscillating and it is reporting correctly. This example doesn't use OpenGL right?  I observe the problem when using RetroArch which does use OGL.

6by9, 2019-03-15:
Either vc4-kms-v3d or vc4-fkms-v3d overlay would do it. ```lsmod | grep vc4``` returning nothing would confirm that you aren't using either.

Hello_fft asks the firmware to run the some code on the QPUs (part of the V3D hardware). 

Suffice to say that it sounds like power saving is kicking in and powering down the V3D and clocks when not in use.

hhromic, 2019-03-15:
Okay here is an interesting observation I just stumbled upon. After running the `hello_fft` program, and testing again RetroArch, the clock does not oscillate anymore either. If I reboot the RPI and run RetroArch, again the clock measurement jumps between 0 and the actual measurement. Then if I run again `hello_fft`, and afterwards RetroArch, the measurements do not jump to 0.

I can confirm that I don't have any vc4 kernel module loaded (with `lsmod`).

hhromic, 2019-03-15:
> Suffice to say that it sounds like power saving is kicking in and powering down the V3D and clocks when not in use.

The GPU is in stress during this test, so I don't think power saving is kicking in here. Sounds more like a bug in clock measurement or gpu clock management in the firmware?

6by9, 2019-03-15:
RetroArch isn't a supported OS, therefore we won't fix anything there.

As I say, it sounds like the V3D block is just dropping into a power saving mode when not doing anything. I'm don't know why it is only being activated at certain times, but in itself it is not a bug.

What is your actual issue with this?

> The GPU is in stress during this test, so I don't think power saving is kicking in here. Sounds more like a bug in clock measurement or gpu clock management in the firmware?

What form of stress test? Are you 100% certain that you are keeping the block constantly busy? How?

hhromic, 2019-03-15:
Just to clarify, RetroArch is not an OS either. It is just an application. I'm using standard Raspbian Lite from the Foundation. There is no usability issue, so no problem in that regard. I can try to find another application to trigger the issue if you are not convinced, no problem either. I want to help.

The actual issue is that measuring the v3d clock with vcgencmd is returning 0-values when the GPU is being used. Maybe this is expected behaviour, in which case nothing is wrong.

> What form of stress test? Are you 100% certain that you are keeping the block constantly busy? How?

As I said, I'm running RetroArch as a graphical application. I will triple-check it is effectively using the GPU and report back to confirm.

However, the fact that running `hello_fft` changes how the clock is reported back by vcgencmd until reboot doesn't mean something might be odd?  Perhaps `hello_fft` leaves something behind in the GPU? Is this application reliable for testing?

hhromic, 2019-03-15:
I found and tried an OpenGLES 2.0 program now. I tested the "tutorial02_red_triangle" from this repository: https://github.com/peepo/openGL-RPi-tutorial.git

The jumpy v3d clock measurements also are observed. So we can discard RetroArch from the picture.

I agree that this application might not be a full-blown stress test for the GPU (it just draws simple objects). However when not using any application the clock is steadily measured as 250Mhz, but when a GPU application is used, it oscillates between 0 (mostly) and 250Mhz.

If you use `hello_fft` once in the running session, the clock measurements don't oscillate anymore until you reboot the machine.

If this is normal/expected, I would like if you can confirm this and then we can close this issue. I just want to make sure the clock measurement for the v3d block are being done correctly.

6by9, 2019-03-15:
OK, I'd misremembered over RetroArch.

There may be something odd left behind, but if there is no functional problem then there's little point expending the effort to investigate.

VideoCore was designed as a mobile graphics chip, therefore the blocks can be controlled fairly finely to save power. Most of that is disabled for the Pi, but some remains.
If you run something like the Taiji GL demo, then the V3D will be powered up and down 60 times a second as rendering a complete frame does not take the whole frame period.
I know the codecs block better, and there the sub-blocks (eg entropy, motion estimation, bitstream parsing, etc) can be powered up and down multple times per frame as and when there is something for them to do.

hello_fft submits a job that requires the QPU solidly for the whole time it is running (computing an FFT). I would have hoped that if power control is active then it powered the V3D block down afterwards, but I'm not too worried if it doesn't.

hhromic, 2019-03-15:
> There may be something odd left behind, but if there is no functional problem then there's little point expending the effort to investigate.
>
> VideoCore was designed as a mobile graphics chip, [...]

Thanks for the details, much appreciated and it makes a lot of sense.

I now found a much better testing program here: https://github.com/mmozeiko/rpi/tree/master/gles2_bcm

It has an option to enable/disable vsync. When vsync is enabled, frames are rendered at ~60fps and when vsync is disabled, frames are rendered at ~1070fps. I guess we can be sure the GPU is stressed with this demo program.

I measured the clock while running that program and indeed the clock becomes much less jumpy (but still goes to 0 sporadically) and way more steady towards 250/300Mhz. So in conclusion I think you are 100% right that the GPU is simply being turned off to save power.

Regarding the `hello_fft` issue, it is just a testing program so I also wouldn't worry much either.

Thanks for taking time to clarify the behaviour of the clock measurements. I hope this issue at least serves to inform for future people finding the same effect.

closed 2019-03-15 by hhromic