raspberrypi/firmware #1124: Second stage PXE ignore DHCP offer in Pi 3B

ronny-nilsson, 2019-03-25:
I am PXE booting a Pi 3B without any SD card or disk. Sometimes the ROM-program successfully downloads a `bootcode.bin`, but then when ROM hands over, the second stage DHCP query fails. I'm working around the network flush bug with dummy broadcasts so that's likely not the issue(?). Plus, the problem is in a recent `bootcode.bin` where I guess the flush bug has been fixed? Only my Pi 3B seems affected though. A parallel Pi 3B+ works.

Here is the second stage serial UART console:
```
Raspberry Pi Bootcode
USB ethernet boot
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
Sending DHCP request
Waiting for dhcp_reply
```
According to the console log bootcode.bin query DHCP endlessly (it never stops). But this is not true. Network querys are only sent for about one minute and then ends, as can be seen below in the DHCP server log. (Verified with tcpdump as well).
```
Mar 25 04:28:54 dhcpd: DHCPDISCOVER from b8:27:eb:5d:5d:74 via eth1
Mar 25 04:28:59 dhcpd: DHCPOFFER on 192.168.1.253 to b8:27:eb:5d:5d:74 via eth1

Mar 25 04:28:59 xinetd[2024]: START: tftp from=192.168.1.253
Mar 25 04:28:59 in.tftpd[31426]: RRQ from 192.168.1.253 filename bootcode.bin
Mar 25 04:28:59 in.tftpd[31427]: RRQ from 192.168.1.253 filename bootsig.bin

Mar 25 04:29:04 dhcpd: DHCPDISCOVER from b8:27:eb:5d:5d:74 via eth1
Mar 25 04:29:04 dhcpd: DHCPOFFER on 192.168.1.253 to b8:27:eb:5d:5d:74 via eth1

Mar 25 04:29:09 dhcpd: DHCPDISCOVER from b8:27:eb:5d:5d:74 via eth1
Mar 25 04:29:09 dhcpd: DHCPOFFER on 192.168.1.253 to b8:27:eb:5d:5d:74 via eth1

Mar 25 04:29:14 dhcpd: DHCPDISCOVER from b8:27:eb:5d:5d:74 via eth1
Mar 25 04:29:14 dhcpd: DHCPOFFER on 192.168.1.253 to b8:27:eb:5d:5d:74 via eth1

Mar 25 04:29:19 dhcpd: DHCPDISCOVER from b8:27:eb:5d:5d:74 via eth1
Mar 25 04:29:19 dhcpd: DHCPOFFER on 192.168.1.253 to b8:27:eb:5d:5d:74 via eth1

Mar 25 04:29:24 dhcpd: DHCPDISCOVER from b8:27:eb:5d:5d:74 via eth1
Mar 25 04:29:24 dhcpd: DHCPOFFER on 192.168.1.253 to b8:27:eb:5d:5d:74 via eth1

Mar 25 04:29:29 dhcpd: DHCPDISCOVER from b8:27:eb:5d:5d:74 via eth1
Mar 25 04:29:29 dhcpd: DHCPOFFER on 192.168.1.253 to b8:27:eb:5d:5d:74 via eth1

Mar 25 04:29:34 dhcpd: DHCPDISCOVER from b8:27:eb:5d:5d:74 via eth1
Mar 25 04:29:34 dhcpd: DHCPOFFER on 192.168.1.253 to b8:27:eb:5d:5d:74 via eth1

Mar 25 04:29:39 dhcpd: DHCPDISCOVER from b8:27:eb:5d:5d:74 via eth1
Mar 25 04:29:39 dhcpd: DHCPOFFER on 192.168.1.253 to b8:27:eb:5d:5d:74 via eth1

Mar 25 04:29:44 dhcpd: DHCPDISCOVER from b8:27:eb:5d:5d:74 via eth1
Mar 25 04:29:44 dhcpd: DHCPOFFER on 192.168.1.253 to b8:27:eb:5d:5d:74 via eth1

Mar 25 04:29:49 dhcpd: DHCPDISCOVER from b8:27:eb:5d:5d:74 via eth1
Mar 25 04:29:49 dhcpd: DHCPOFFER on 192.168.1.253 to b8:27:eb:5d:5d:74 via eth1

Mar 25 04:29:54 dhcpd: DHCPDISCOVER from b8:27:eb:5d:5d:74 via eth1
Mar 25 04:29:54 dhcpd: DHCPOFFER on 192.168.1.253 to b8:27:eb:5d:5d:74 via eth1
```

Versions used:
- bootcode.bin: 1.20190215  (98997f363e3683ead4f50c37902169248628303a)
- ISC DHCP server: v3
- Pi 3B, board revision a02082

maxnet, 2019-04-04:
```
**Mar 25 04:29:04** dhcpd: DHCPDISCOVER from b8:27:eb:5d:5d:74 via eth1
**Mar 25 04:29:04** dhcpd: DHCPOFFER on 192.168.1.253 to b8:27:eb:5d:5d:74 via eth1
```

There were problems with instant replies in the past as well.
So suggest you try if it works better with dnsmasq as DHCP server, and dhcp-reply-delay=2

ronny-nilsson, 2019-04-04:
Yes, I know there has been problems with instant replies. This is why the very first DHCP reply in my log above has a five second delay. The second reply is instant - but then latest `bootcode.bin` has already started. And I assume it has been fixed there.