raspberrypi/firmware #1126: CM3+ «Add bridge failed, Packard not installed»

EspenT, 2019-03-28:
Hi I know I’m probably doing something wrong here, and need some help. I’ve got a Raspian stretch image running perfectly on a Pi 3 B+

Transferred it to a new CM 3+ and it is stuck at the rainbow screen. 
Copied all files from the master branch /firmware to my boot partition and it does both but no network. Apparently bridging is needed and not enabled in the kernel.

ESandM-Public, 2020-01-16:
Similar boat....currently trying to UART into a CLI, but realizing it might be a combination of a firmware issue AND initial boot config, I thought it might be more straightforward to just re-flash the eMMC with an image that addressed these issues directly.

If you resolved, can you share how?

EspenT, 2020-01-16:
Unfortunately I can not remember how I solved this. Should have put it up here while I was stil remembering. What I can tell you is that the Mass storage mode will give you access to all files. Including boot files.

ESandM-Public, 2020-01-16:
Ok, that's good news for me.  Thanks so much.  My power supply is a
switched, 2 A rating, so I'm assuming I either need to swap out a file, or
just re-flash the eMMC with an updated image.

Thanks much EspenT for your timely reply!

Best,
Dan

-- 
*Daniel M. Drendall*
*Power Marketer/Trader*
*Midland Cogeneration Venture*
*100 Progress Place*
*Midland, MI 48640*
*Tel:  989/633-7850*
*Mob:  989/860-4384*
*Fax:  989/633-7857*
*dmdrendall@midcogen.com <dmdrendall@midcogen.com>*
*IM:  dmdrendall*