raspberrypi/firmware #1127: Remove Video scheduler and Clock elements from hello_pi/hello_video/video.c to improve latency

ktb92677, 2019-04-02:
As the title suggests, I am trying to remove video scheduler and clock elements form hello_video/video.c and I am having a lot of trouble getting it to work. My code is below:

```
void Initialize_OPENMAX_H264_DECODER(OPENMAX_H264_DECODER* decoder) {
  decoder->port_settings_changed = 0;
  decoder->first_packet          = 1;
  decoder->Buffer = NULL;
  memset(decoder->List,    0, sizeof(decoder->List));
  memset(decoder->Tunnel,  0, sizeof(decoder->Tunnel));
  memset(&decoder->Format, 0, sizeof(OMX_VIDEO_PARAM_PORTFORMATTYPE));

  bcm_host_init();

  decoder->Display = vc_dispmanx_display_open(0); //why is this zero? who knows...

  if (vc_dispmanx_display_get_info(decoder->Display, &decoder->Info)) {
    return;
  }

  decoder->Width  = decoder->Info.width;
  decoder->Height = decoder->Info.height;

  if((decoder->Client = ilclient_init()) == NULL) {
     return;
  }
  if (OMX_Init() != OMX_ErrorNone) {
     ilclient_destroy(decoder->Client);
     return;
  }
  if(ilclient_create_component(decoder->Client, &decoder->Video_Decode, "video_decode", ILCLIENT_CREATE_FLAGS_T(ILCLIENT_DISABLE_ALL_PORTS | ILCLIENT_ENABLE_INPUT_BUFFERS)) != 0) {
    return;
  }
  decoder->List[0] = decoder->Video_Decode;

  if (ilclient_create_component(decoder->Client, &decoder->Video_Render, "video_render", ILCLIENT_DISABLE_ALL_PORTS) != 0) {
    return;
  }
  decoder->List[1] = decoder->Video_Render;

	
  set_tunnel(decoder->Tunnel,     decoder->Video_Decode,    131, decoder->Video_Render, 10);

  ilclient_change_component_state(decoder->Video_Decode, OMX_StateExecuting);

  decoder->Format.nSize              = sizeof(OMX_VIDEO_PARAM_PORTFORMATTYPE);
  decoder->Format.nVersion.nVersion  = OMX_VERSION;
  decoder->Format.nPortIndex         = 130;
  decoder->Format.eCompressionFormat = OMX_VIDEO_CodingAVC;

  if (OMX_SetParameter(ILC_GET_HANDLE(decoder->Video_Decode), OMX_IndexParamVideoPortFormat, &decoder->Format) == OMX_ErrorNone &&
        ilclient_enable_port_buffers(decoder->Video_Decode, 130, NULL, NULL, NULL) == 0) {
     ilclient_change_component_state(decoder->Video_Decode, OMX_StateExecuting);
  }

}

void Render_H264_Frame(OPENMAX_H264_DECODER* decoder, char* buffer, int size) {
  int process_size = 0;
  while (process_size < size) {
    if ((decoder->Buffer = ilclient_get_input_buffer(decoder->Video_Decode, 130, 1)) != NULL) {
       // feed data and wait until we get port settings changed
       unsigned char *dest = decoder->Buffer->pBuffer;

       copy(buffer + process_size, buffer + process_size + (((size - process_size) > decoder->Buffer->nAllocLen) ? decoder->Buffer->nAllocLen : (size - process_size)), dest);

       if (decoder->port_settings_changed == 0) {
         if (ilclient_remove_event(decoder->Video_Decode, OMX_EventPortSettingsChanged, 131, 0, 0, 1) == 0) {
           decoder->port_settings_changed = 1;

           if(ilclient_setup_tunnel(decoder->Tunnel, 0, 0) != 0) {
              return;
           }

           ilclient_change_component_state(decoder->Video_Render, OMX_StateExecuting);
         }
       }

       decoder->Buffer->nFilledLen = (((size - process_size) > decoder->Buffer->nAllocLen) ? decoder->Buffer->nAllocLen : (size - process_size));

       process_size += (((size - process_size) > decoder->Buffer->nAllocLen) ? decoder->Buffer->nAllocLen : (size - process_size));

       decoder->Buffer->nOffset = 0;
       if(decoder->first_packet) {
          decoder->Buffer->nFlags = OMX_BUFFERFLAG_STARTTIME;
          decoder->first_packet = 0;
       }
       else
          decoder->Buffer->nFlags = OMX_BUFFERFLAG_TIME_UNKNOWN;

       if (OMX_EmptyThisBuffer(ILC_GET_HANDLE(decoder->Video_Decode), decoder->Buffer) != OMX_ErrorNone) {
          return;
       }
    }
  }
}
```

The system is able to initialize but there is nothing displayed on the screen. Any help would be greatly appreciated!

6by9, 2019-04-02:
This is really an issue for https://github.com/raspberrypi/userland/issues, as that is where the source for hello_video originates.

Secondly your code appears to be a significant change from hello_video as things like the function names have changed. Sorry, I'm not going to debug your code.
I've made the required changes to hello_video and pushed them to https://github.com/6by9/userland/tree/fw1127 as a quick hack.

pelwell, 2019-04-02:
As this is really a question rather than an issue, the Forums are a better home for your query - try the [Graphics](https://www.raspberrypi.org/forums/viewforum.php?f=67) section.

6by9, 2019-04-02:
> As this is really a question rather than an issue, the Forums are a better home for your query - try the Graphics section.

I now see they already have - https://www.raspberrypi.org/forums/viewtopic.php?f=70&t=237236

Please don't open issues if you've asked on the forum and have been too impatient to wait for an answer.
@pelwell please close this.

closed 2019-04-02 by pelwell